OrcaSlicer/OrcaSlicer #2842: Internal bridge uncesseary

VpopUnik, 2023-11-22:
### OrcaSlicer Version

1.8

### OS version

Win 10

### Printer

SV06 0.4nozzle

### How to reproduce

This issue is on several models that have a top surface. The Orca cube for example

### Actual results

Not sure if this is a bug or working as intended. If it is as expected, this could be improved(?)
Over the infill, the top layer is created as internal bridge and applies bridge settings such as slower printing. Even If I add 99% infill and 10 top layers, the first solid layer is considered a bridge. This should be a top solid layer.

Posted the issue here:
https://www.facebook.com/groups/orcaslicerusers/permalink/624935016513830/

### Expected results

Instead of a internal bridge the layer should adapt to be a solid top layer if its not to thin.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-11-22:
This is not a bug but rather a feature. The first layer over sparse infill is indeed a bridge - it extrudes partly supported so lower speeds and bridging flow settings need to apply to get good quality printing on the layers above it. You can control the print speed of the internal bridge independently of the external bridge - setting can be found under Overhang speed -> bridge -> External. That way you can set speed for it if you dont need it to slow down.

killerline10168, 2023-11-22:
If you don't like internal bridges, you can always turn them off. 

As this is an optional feature and adaptable depending on the loaded model, this ticket can probably be closed.

SoftFever, 2023-11-22:
> Internal bridging really is needed, especially with low layer height/higher printing speed. The cases when it isn't correctly applied by the slicer demonstrate how terrible bridging over sparse infill with full speed (e.g. 350 instead of 75 mm/s) works. The mess it creates (lumps, strings, loose ends sticking out of the print) propagates upwards and still causes the nozzle to hit the print many layers higher, resulting in visible surface bumps.
> 
> 
> 
> Slightly off-topic rant: I wish the first layer over sparse infill actually always was an internal bridge. Sadly it's not, see https://github.com/SoftFever/OrcaSlicer/issues/636 (or https://github.com/bambulab/BambuStudio/issues/1653). 
> 
> 
> 
> <details>
> 
> <summary>Pictures</summary>
> 
> 
> 
> [image]
> 
> 
> 
> [image]
> 
> 
> 
> </details>
> 
> 

You can slow down the speed for internal bridges in Orca.

discip, 2023-11-23:
@VpopUnik
After being advised by @igiannakas how to solve your issue, please consider closing it. 😊

thanks

VpopUnik, 2023-11-24:
Thanks for all your replies. I set internal bridge to 100mms instead of 150% of external (25mms). Seemed to work just fine.

closed 2023-11-24 by VpopUnik

jackmiller1986, 2025-03-21:
An option to take it complete of would be recommend, that also works in cura , there are multi objects that wont need that , and it would be good to had an feature to take it complete off to some prints, that option are nowhere.