OrcaSlicer/OrcaSlicer #2845: Bridge Infill not generated on bridge features when using Color Painting tools

blownupp, 2023-11-22:
### OrcaSlicer Version

1.8.0

### OS version

Windows 10.0.19045

### Additional system information

CPU: 11th Gen Intel Core i7-11850
GPU: NVIDIA RTX A2000

### Printer

Bambu Lab X1C

### How to reproduce

1. Open an STL/3MF/STEP file with bridge features
2. Use any of the "Color painting" tools (besides "Height Range") on a bridged feature
3. Click "Slice plate"

### Actual results

The bridged features that have been painted a different color are sliced with the number of selected walls as "Overhang wall" instead of as all "Bridge" or "Internal Bridge".

[image]

[image]

If using the "Height range" tool, this bug is not present, and bridge features are sliced with "Bridge" or "Internal Bridge" settings instead of "Overhang wall".

### Expected results

Bridge features should be sliced with "Bridge" or "Internal Bridge" settings instead of "Overhang wall" settings. When sliced as "Overhang wall" most of the generated walls/perimeters are printed over thin air (this depends on the "Quality" -> "Advanced" -> "Order of inner wall/outer wall/infill" setting - the default setting is for the inner walls to be printed before the outer walls), when a "Bridge" or "Internal Bridge" would be anchored to a perimeter/wall on the previous layer:

[image]

[image]

[image]

### Project file & Debug log uploads

[Orca_overhang-wall-bridge_bug.zip](https://github.com/SoftFever/OrcaSlicer/files/13442432/Orca_overhang-wall-bridge_bug.zip)
[debug_Wed_Nov_22_11_38_07_57216.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13442433/debug_Wed_Nov_22_11_38_07_57216.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mrmultifunk, 2023-12-18:
https://github.com/SoftFever/OrcaSlicer/assets/56545669/71ece4a5-2fbc-4ef7-a0c2-c72dcef4506a

I have the exact same issue. Did you find any solution?

blownupp, 2023-12-19:
> Orca.issue.mp4 
> 
> I have the exact same issue. Did you find any solution?

I have not found a solution - we can only hope this pops up on a developers radar or more people submit similar issues.

mrmultifunk, 2023-12-19:
I hope so, it is pretty frustrating. it seems that not many people are affected by this, so i wonder if we have a setting somewhere that can cause this 🤔

blownupp, 2023-12-22:
> I hope so, it is pretty frustrating. it seems that not many people are affected by this, so i wonder if we have a setting somewhere that can cause this 🤔

There’s that, or it’s a very specific use case where it becomes apparent - in our case the color change is only where the bridging is, most people doing color changes probably aren’t doing them specifically where there’s bridging.

I think this issue is related to another one actually and has to do with how perimeters are handled - it’s like when there’s a color change there’s something internally that’s “reset” and the perimeter generator goes to start regenerating perimeters first instead of looking for any other features that should override it (perimeters are generated first, and they’re always the outermost area generated - when there’s a filament change, locally the slicer thinks “oh I need to generate the given number of perimeters before I do anything else”)

VadymMakhonin, 2024-07-10:
Are there any updates on this issue? I encountered the same problem, the color changes just in the air.

andreaspolydorides, 2024-12-27:
Encountered the same issue in orcaslicer 2.2.0 where coloured bridging is prited on air. Desired outcome would be for bridging to expand further into infill for support.

https://github.com/user-attachments/assets/e0ee4add-a109-489a-bf77-9c91b47e3bbb

closed 2026-11-10 by RF47