OrcaSlicer/OrcaSlicer #2851: Failed processing of the filename_format template

Kalougear, 2023-11-22:
### OrcaSlicer Version

1.8

### OS version

Windows 10

### Printer

Ender 3v2 Modified "klipper"

### How to reproduce

I just tried to slice a model,then try to save it,I tried it with 10 models....when simultaneously slicing them perfect in any other slicer.

### Actual results

I get then this error when im trying to just save it. 
"Failed processing of the filename_format template."
[image]

### Expected results

The result i want is just to slice and save it.

### Project file & Debug log uploads

There is nothing in logs.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Nbeknel, 2023-11-23:
Have you tried using "{filament_type[0]}" or "{filament_type[initial_tool]}" instead of "{filament_type{0}}"? Notice the square brackets instead of the curly brackets.

closed 2024-03-01 by github-actions[bot]

Gwaboo, 2024-03-02:
{filament_preset[0]}

ad5muser, 2024-03-17:
> Have you tried using "{filament_type[0]}" or "{filament_type[initial_tool]}" instead of "{filament_type{0}}"? Notice the square brackets instead of the curly brackets.

Hi sir, I am having a similar issue. Can you please explain how to do what you are suggesting?