OrcaSlicer/OrcaSlicer #2852: SimplyPrint Integration

ETS5, 2023-11-23:
**Which printers will be beneficial to this feature**
All printers supported by Simplyprint

**Describe the solution you'd like**
I was wondering if their could be some collaboration from simplyprint and yall to see if there was a way to be able to integrate Ocraslicer into Simplyprint.

**Describe alternatives you've considered**
I can yes just slice and export the gcode and then upload to Simplyprint it would be awesome to just upload it directly to there.

Ocraftyone, 2023-12-10:
Documentation regarding integrations: https://simplyprint.io/integrations/

closed 2024-03-18 by github-actions[bot]

AlbertMN, 2024-03-30:
@ETS5 this is now live :) https://github.com/SoftFever/OrcaSlicer/pull/4525