OrcaSlicer/OrcaSlicer #2860: orcaslicer starts with empty setup window

didli, 2023-11-23:
### OrcaSlicer Version

1.8.0

### OS version

Fedora release 38 (Thirty Eight)

### Additional system information

Kernel: 6.5.10-200.fc38.x86_64 arch: x86_64 / Desktop: Xfce v: 4.18.1 / Distro: Fedora release 38 (Thirty Eight)
Graphics : NVIDIA TU106 [GeForce RTX 2060 Rev. A] driver: nvidia v: 545.29.02

### Printer

Ender 3 v2 Neo

### How to reproduce

# running the appimage directly from terminal : 
- some critical / warnings from theme / gtk window
`[2023-11-23 15:37:02.601964] [0x00007f605e93be80] [trace]   Initializing StaticPrintConfigs
(orca-slicer:1637404): Gtk-CRITICAL **: 15:37:05.551: gtk_window_resize: assertion 'width > 0' failed
(orca-slicer:1637404): GLib-GObject-CRITICAL **: 15:37:05.575: invalid cast from 'wxPizza' to 'GtkCellLayout'
(...)`
- then these errors as the windows are loaded :
`KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1198x713: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1198x713: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1198x713: Permission denied
Failed to create EGL images for DMABufs with file descriptors -1, -1 and -1`

### Actual results

orcaslicer starts with a certificate issue : 
- - - 
[image]
- - - 
then, global window and setup window shows up : 
- - -
[image]
- - -  

### Expected results

While I could understand its concerns, I expect orcaslicer to stop hating me and my machine.

### Project file & Debug log uploads

1st run log file :
[debug_Thu_Nov_23_15_21_38_1622907.log](https://github.com/SoftFever/OrcaSlicer/files/13451352/debug_Thu_Nov_23_15_21_38_1622907.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

etrigan63, 2023-11-23:
Same thing happening in Arch Linux.

yomgui1, 2023-12-17:
Same after updated my NVidia drivers to 535.146.02

yomgui1, 2023-12-17:
After a quick search, it looks to be an issue with WebKitGTK and DMA buffer.
Starting Orca with following environment variable set resolve the display issue: WEBKIT_DISABLE_DMABUF_RENDERER=1

closed 2024-03-24 by github-actions[bot]

didli, 2024-05-01:
> Starting Orca with following environment variable set, resolves this display issue: WEBKIT_DISABLE_DMABUF_RENDERER=1
Works fine for me with this env set as mentioned. However as the issue is still in-going (at least with latest stable 2.0 version), this only serves as a workaround.

letsruletheworld, 2024-10-20:
only helps with the startupwindow. the 3D view is still missing.