OrcaSlicer/OrcaSlicer #2867: Unable to turn of heated bed

tlhintoq, 2023-11-23:
### OrcaSlicer Version

1.7

### OS version

Windows10x64

### Printer

CR10s-5 (but same with all)

### How to reproduce

1. Comment out the bed heating commands of the machine gCode, in the machine configuration.
2. Slice a file
3. Notice that there is a bed heating command in the `EXECUTABLE_BLOCK_START` just above the `CUSTOM` block.
[image]

### Actual results

3. Notice that there is a bed heating command in the `EXECUTABLE_BLOCK_START` just above the `CUSTOM` block.

This makes it impossible to configure a machine that has an independently controlled bed heater such as an AC powered heater with its own controller like Keenovo bed heater.

The material should have a bed temperature.  That material might be used on several machines in the same shop.  Whether or not a specific machine has a heated bed should be configurable in the machine setup and not be reliant on the material definition.

### Expected results

Bed heating is already in the custom code block.  It should not be forced in an additional spot.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2023-11-25:
Anti-bot auto-close nudge remark

tlhintoq, 2023-12-03:
Hey bot - Don't close me now

tlhintoq, 2023-12-06:
Keep alive comment

tlhintoq, 2023-12-08:
FYI: Still a problem in 1.8.1
[image]

tlhintoq, 2023-12-14:
Keep alive

tlhintoq, 2023-12-23:
Tag - you're it - Still an open issue.

tlhintoq, 2023-12-30:
Still an issue on 1.9.0 beta

tlhintoq, 2024-01-17:
Keep alive - Still an issue in 2.0 developer/nightly
[image]

tlhintoq, 2024-01-22:
Interesting observation for nightly 20241221

If you select a temperature tower the generated gcode doesn't include setting the bed temp before the custom block.  It relies on the custom block to do its job.

But if you have another model instead... Even a built-in item like the Orca cube... Then it sets the bed temp.

@SoftFever @Noisyfox  Really would be nice to get this fixed for those of us with externally controlled AC heated beds.  Just seems like a fix that can't be more than 20 minutes to track down where the bed temp is being injected when it doesn't need to be and the user can't block it.

[image]

tlhintoq, 2024-02-08:
Hey bot, please don't close me - this is still an issue.

tlhintoq, 2024-02-24:
@SoftFever @Noisyfox 
Just wondering...
Since fixing this seems to be a long-term issue...
Is there a gcode change ability that I can attach to jobs for the s5?
Like
` replace m190 with ;m190`
but have it happen just on jobs sliced for the printer with no heat bed.
This bit of having to open every s5 job in Notepad++, do a find, delete the line that should be there anyway,... Its getting old.

tlhintoq, 2024-02-24:
> @tlhintoq Orca will automatically insert heating command if no heating command is detected in custom gcode. Not sure if it's a bug in your case. 

That is absolutely a/the bug in this case. That is the behavior I'm trying to stop.  The printer has no heated bed controlled by the printer controller. The heated bed is plugged directly to AC mains power and has its own controller.
[image]

There is no bed thermistor connected to the printer. If you try to send a command to heat the bed it squeals, throws up an error on the LCD. "Min temp error"

So anyone with an externally controlled heatbed can't use Orca.  That's at least half of the CR10s-4 and CR10s-5 users I know on those groups. Glass beds 400-500-600mm big take forever to heat off a 24vdc printer power supply: That's why owners of big bed machines get the AC mains powered silicone heat beds. 

They are forced to use Cura over a simple matter of a checkbox on the printer configuration page.
[image]
And a simple line of code _to the effect of_ this logi:
`if {HasHeatedBed} then InsertBedHeatComman(BedTemp);`
however the actual code in Orca would follow that logic.

Does that help understand the problem better?

tlhintoq, 2024-03-13:
Problem still exists in current 2.0 nightly as of 13mar2024

tlhintoq, 2024-04-15:
Keep alive.  Problem still exists in current 2.0 nightly as of 15apr2024

tlhintoq, 2024-05-10:
Bot nudge.

closed 2024-08-17 by github-actions[bot]

jltx1, 2025-05-02:
this is absolutely wrong.  A slicer should NEVER control the machine without my consent.  I use macros to manage heaters and Orca decides to insert its own heating commands.  This must be allowed to be disabled.  If someone omits a heating command in their start g-code then that is on them.  Don't try to be a nanny.  Please fix.

umartechboy, 2025-07-05:
Try 
M190 S0
In your start G code. Orca, just like cura, will skip its own heating command.