OrcaSlicer/OrcaSlicer #2869: Custom Overrides

alfondehesa, 2023-11-24:
**Describe the solution you'd like**
Add custom setting overrides to filament/printer as desired. Every setting should be available to override for every preset (filament printer and process).

For example:

1. Z Offset is usually defined in the printer preset. But maybe I like to squeeze ABS down just a bit more than other filaments. 

2. Maybe I want ABS to set any process profile to have brims enabled by default (and override the process regardless).

3. Maybe I have a specific printer that is in an enclosure and I want to override cooling settings regardless of filament from there. This would be super cool.

alfondehesa, 2023-11-24:
In terms of establishing a hierarchy, I think for every preset (printer filament and process) there should be a tab, or clone of settings, called "Overrides" (something like "Top surface pattern Override" or "Z Offset Override" or things like this). If no other preset conflicts with the override, then it would be straight forward. If there are two conflicting overrides (i.e. the printer says 0 Z offset override but the filament says -0.2 Z offset override), then it should probably be prioritized as: printer-filament-process.

This is because when setting printer settings you are usually specifying hard constraints (especially if they are overrides), filaments closer to "nice to haves", and process closer to "delighters".

closed 2024-03-01 by github-actions[bot]

dprossner, 2025-06-22:
The ability to set custom setting overrides would help avoid so many headaches. In my case, the retraction length on tool change varies significantly for different materials. While the normal retraction length can be overridden per filament, the retraction length before each tool change has to be set in the printer profile under each extruder. I want to be able to have TPU always retract at my desired length, whether it's mid-layer, before layer change, or before tool change, and not have to keep changing my whole printer config each time I switch between TPU and PLA. The current limits on options for settings overrides cause so much friction and I imagine many headaches would be resolved by the ability to add custom setting overrides to each filament and/or printer profile