OrcaSlicer/OrcaSlicer #2872: Arrange function broken

altmenorg, 2023-11-24:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11 / Ubuntu

### Printer

FLSun V400

### How to reproduce

Place multiple objects on the place, and try to arrange...
[image]

### Actual results

Tried to both arrange "all plates" or arrange on plate (A and Shift + A) and got the following results
[image]
[image]

### Expected results

Should look something like this :
[image]

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13457665/log.zip)
[test.zip](https://github.com/SoftFever/OrcaSlicer/files/13457672/test.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

julie777, 2023-12-10:
I have the same issue. My case is very simple. With 6 objects on the plate that will easily fit. 
[image]
I click on the plate arrange icon and auto-arrange moves one of the items off the plate.
[image]

altmenorg, 2024-01-24:
Another weird effect of the issue... I tried the "fill bed with copies" function. The issue is the same.
Here is the plate before
[image]

Here is the plate after
[image]

Doesn't fill at all. It seems like the geometry of my plate isn't right from Orcaslicer perspective...

Everything works like a charm with Prusaslicer, by the way
[image]

I only own a delta printer. Can someone confirm if the "fill bed with instances" works with a more standard bed geometry ?

kleinwareio, 2024-02-22:
I have this same issue. v1.9.1 and a 300mm diameter build plate (v400).

From the outside, it appears like the algorithm is able to correctly place the first column of parts, but then its idea of what the bounding box is screws up for the remaining columns.

The auto-arrange appears to break in the same way the "[Fill bed with copies](https://github.com/SoftFever/OrcaSlicer/issues/1904)" breaks. I suspect they're using the same algorithm under the hood.

The bed was populated by "Fill bed...". Then cloning one more item and moving it onto an empty area of the bed and then performing auto-arrange ends up moving that newly added item off the bed to the "outside."

[image]

closed 2024-06-01 by github-actions[bot]