OrcaSlicer/OrcaSlicer #2877: 'Convert holes to polyholes' setting turns some holes into octagons

kozross, 2023-11-24:
### OrcaSlicer Version

1.8.0

### OS version

Arch Linux

### Additional system information

AMD Ryzen 9 5900X (probably doesn't matter)

### Printer

Qidi XPlus-3

### How to reproduce

1. Open the project attached.
2. Attempt to slice, and look at the first layer.
3. Turn off 'Convert holes to polyholes', then slice again.

### Actual results

[image]

With polyholes turned off, it instead looks right:

[image]

### Expected results

Polyholes shouldn't be turning those holes into polygons, especially given that it handles other circular holes normally.

### Project file & Debug log uploads

[neko3s-plus-new-y-rod-holder-front.zip](https://github.com/SoftFever/OrcaSlicer/files/13462326/neko3s-plus-new-y-rod-holder-front.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DVSVIDEO, 2023-12-01:
Opening a wrong ticket because of something you can't use? It's funny...
For a long time I didn't know what "Polyhole" was either, so I didn't turn it on.
Then I started looking into it. I had Google handy and found out what it was.
I suggest you read up and inform yourself.
I read somewhere and it's very true: all the information in the world is on the Internet. In the future, we'll call a smart person the one who finds what he needs....

jasonslay, 2023-12-23:
Would making **Convert holes to polyholes** a link to the URL that's in the hover text help add clarity here? I was also a bit confused at first and manually searched instead of transposing the URL (which isn't even possible when you move your mouse away).

closed 2024-03-23 by kozross