OrcaSlicer/OrcaSlicer #2878: Disable / Enable Drop to build plate for one extruder multicolor prints

sismo99, 2023-11-25:
It is not possible to raise objects above the printing bed, which is necessary to be able to print in multiple colors with a single extruder. You cannot even load objects that already have the elevation from the design and are lowered to the bed.

**Which printers will be beneficial to this feature**
All printers with only one extruder 

**Describe the solution you'd like**
Add the option to disable objects from being automatically brought to the print bed and allow objects to be elevated above the print bed

**Describe alternatives you've considered**
Just add the option to allow objects to be in the air without necessarily having to be stuck to the plate I prefer OrcaSlicer a thousand times but Cura does have the option I hope you can implement it

**Additional context**
[image]
[image]

joebb3, 2024-01-04:
Concur.  It would be nice to be able to join multi piece models and merge them directly in Orca, which is exceedingly difficult when pieces snap to the build plate.

soundtechscott, 2024-02-09:
yes... I need to be able to float an object and use supports instead of snapping to the bed
can we please disable the print snapping to the bed automatically

muddledog, 2024-03-26:
Agree - this would be a very nice addition.

vgdh, 2024-03-26:
when you group them into assemble you able to move parts anywhere you want while one part of the assemble remaining in touch  with the build plate.

soundtechscott, 2024-03-26:
> when you group them into assemble you able to move parts anywhere you want while one part of the assemble remaining in touch with the build plate.

you are missing the point.....
there is nothing to group with a single object/single file

and the software still requires something to be touching the buildplate

make it so you can float the object
then you can use supports under an object and it doesnt need to touch the build plate....sometimes we dont want the build plate texture on the underside...only way to do this is to float it and use supports underneath instead...

you can do this in other software....
if orca is supposed to be so good, then this is a feature that is needed
lacking this feature invalidates doing several of my regular prints using orca.

Gozania, 2024-04-11:
+1 one for this.  I have zero CAD skills and make my needed parts within the slicer since all I need are simple shapes to make the parts I need.  Let us float objects to be merged with the main part after it is in the correct position.

b1tfan, 2024-06-03:
please include, same for me for manual multicolor prints

AJolly, 2024-06-06:
Would be very useful

KickingKing, 2024-06-20:
I am amazed that it is not included in Orca.
It is in all other slicers for more than 10 years now.

kitkat0981, 2024-06-24:
This is a must have. Baffled that it’s not even an option now…

devoidv3, 2024-06-24:
This would be absolutely amazing to have!!!!

Galaxi-JonnyStepps, 2024-07-05:
I want to chime and say that I specifically need this exact feature to join and assemble 2 objects together but it is currently not possible to do unless I do some janky work around by adding a cube that's the desired height and gluing it to the bottom of the object in the slicer.

Galaxi-JonnyStepps, 2024-07-05:
> when you group them into assemble you able to move parts anywhere you want while one part of the assemble remaining in touch with the build plate.

This is incorrect, you cannot move objects after they are placed in assemble, I have 2 objects right now and when I click "assemble" it binds them together and you are not able to move the objects separately.

discip, 2024-07-17:
@Galaxi-JonnyStepps
> **This is incorrect,** you cannot move objects after they are placed in assemble, I have 2 objects right now and when I click "assemble" it binds them together and you are not able to move the objects separately.

This is incorrect, you can move objects after assembly if you click not on the object itself but on its name. 😊.

See:

[image]

discip, 2024-07-17:
@Galaxi-JonnyStepps
> **This is incorrect,** you cannot move objects after they are placed in assemble, I have 2 objects right now and when I click "assemble" it binds them together and you are not able to move the objects separately.

This is incorrect, you can move objects after assembly if you click not on the object itself but on its name 😊.

See:

[image]

quantumfrost, 2024-09-08:
Yes, this should be an option rather than forcing us to do janky workarounds. Just display a warning if something is off the buildplate and let the user decide.

silvernull, 2024-10-27:
+ 1

closed 2025-02-02 by github-actions[bot]

quantumfrost, 2025-02-02:
Issues really shouldn't be closed just because they are stale. There has been no activity in 90 days because there isn't much more to add to the discussion until developers take a look.

primesun, 2025-02-08:
Same feature request for Bambu Slicer is here: https://github.com/bambulab/BambuStudio/issues/3495

0o0will0o0, 2025-07-21:
It appears that this bug is being intentionally ignored by the developers for an unknown reason