OrcaSlicer/OrcaSlicer #2879: Turkish Language file

olcayoren, 2023-11-25:
### OrcaSlicer Version

1.8.0

### OS version

Macos Sonoma 14.0

### Additional system information

Macbook m1

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

Hello, I could not find the lines in the Turkish language file (OrcaSlicer_tr.po) to edit the English fields in the picture? How can I edit it?

### Actual results

Although Turkish is selected as the language, some fields on the login screen are in English.
[image]

### Expected results

Under normal circumstances, all terms are expected to be in Turkish.

### Project file & Debug log uploads

No need for log file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

olcayoren, 2023-12-08:
Hello, I could not see the corresponding values ​​in the language file, how should they be added? @Noisyfox

Noisyfox, 2023-12-08:
Those are placed in the resource/web folder

closed 2023-12-09 by olcayoren