OrcaSlicer/OrcaSlicer #2880: build fails on MacOS 14.1.1 / Xcode 15.0.1

superfloh247, 2023-11-25:
### OrcaSlicer Version

main a6163a1

### OS version

MacOS 14.1.1

### Additional system information

AppleSilicon M1pro
Xcode 15.0.1

### Printer

Bambu P1P

### How to reproduce

git clone --depth=1 https://github.com/SoftFever/OrcaSlicer.git
cd OrcaSlicer
./build_release_macos.sh 

### Actual results

build fails

### Expected results

build succeeds

### Project file & Debug log uploads

[build.log](https://github.com/SoftFever/OrcaSlicer/files/13464209/build.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Ocraftyone, 2023-12-10:
Have you tried again recently. Wx got a major update.

superfloh247, 2023-12-10:
yes, still fails w/o my patch to CMakeLists.txt, see above

eoyilmaz, 2024-01-16:
> partly fixed with [ce9a8d7](https://github.com/SoftFever/OrcaSlicer/commit/ce9a8d7b20b336319bec7c79554ca24192e38d94)
> 
> just one patch needed
> 
> ```
> --- a/CMakeLists.txt
> +++ b/CMakeLists.txt
> @@ -52,7 +52,7 @@ if (APPLE)
>      if (CMAKE_MACOSX_BUNDLE)
>          set(CMAKE_INSTALL_RPATH @executable_path/../Frameworks)
>      endif()
> -    set(CMAKE_OSX_DEPLOYMENT_TARGET "11.3" CACHE STRING "Minimum OS X deployment version" FORCE)
> +    set(CMAKE_OSX_DEPLOYMENT_TARGET "14.0" CACHE STRING "Minimum OS X deployment version" FORCE)
>      SET(CMAKE_XCODE_ATTRIBUTE_PRODUCT_BUNDLE_IDENTIFIER "com.softfever3d.orca-slicer")
>  
>  endif ()
> ```
> 
> with that applied to CMakeLists.txt the build succeeds

your suggestion is not working for me.

@SoftFever this is a huge blocker for me, I want to contribute to the project, but can't build it to check my code...

eoyilmaz, 2024-01-16:
Creating a symbolic link to `MacOSX14.0.sdk -> MacOSX14.2.sdk` at `/Library/Developer/CommandLineTools/SDKs` appears to be working... started building now...

eoyilmaz, 2024-01-16:
Oh, just realized that I can pass the XCode target version to the build script with:

`build_release_macos.sh -t 14.2`

So I think there is no need to modify the `CMakelists.txt` files...

eoyilmaz, 2024-01-16:
No it's not buiding:

```shell
eoyilmaz@Erkans-MBP OrcaSlicer % ./build_release_macos.sh -a arm64 -t 14.2 -s
Arch: arm64
BUILD_TARGET: slicer
OSX_DEPLOYMENT_TARGET: 14.2
building slicer...
CMake Error:
  Xcode 1.5 not supported.
```

eoyilmaz, 2024-01-16:
I think I don't have XCode installed and it is defaulting back to the oldest one, which is version `1.5` and complaining about that... and there is nothing wrong with the build scripts...

eoyilmaz, 2024-01-16:
Yeah that was the issue, and I'm able to build it without any issues using:

`./build_release_macos.sh -a arm64 -t 14.2`

SoftFever, 2024-01-16:
Yeah, Apple makes compiling C++ on their OS unnecessarily complex and painful for no apparent reason.

SoftFever, 2024-01-16:
Looks like the problem solved

closed 2024-01-16 by SoftFever