OrcaSlicer/OrcaSlicer #2884: Random Large Circles while printing but not shown on slicer

mmorning, 2023-11-25:
### OrcaSlicer Version

1.8.0

### OS version

Windows 10 Home

### Additional system information

Processor: Intel(R) Core(TM) i7-3820QM CPU @ 2.70GHz   2.70 GHz
Ram: 24.0 GB
Graphics: Nvidia GeForce GTX 675M

### Printer

Elegoo Neptune 2s Custom Duel Extruder

### How to reproduce

This can only be produces while printing. It happened on this file and a few others. I believe it is related to mouse ears. I noticed I had it enabled. But it doesn't show up on any specific layers. the first time it happened it printed about layer 54. The next time I printed it happened layer 4. 

### Actual results

It will create a large circle. in different locations. I however did not get a screen shot but I did provide the sliced code. With these large circles it causes the printer to try and print the object out of bounds and will cause a layer shift. Which causes the print to fail. 

### Expected results

For the circles to show on the previewed slice and also for it not to print the circles. I am running the print again to see if disabling mouse ears under Brim Type fixes the issue. 

### Project file & Debug log uploads

[Flexi Factory Bambu lab Panda_Patreon_0.12mm_PLA_Elegoo Neptune 2D_1d9h40m.zip](https://github.com/SoftFever/OrcaSlicer/files/13464525/Flexi.Factory.Bambu.lab.Panda_Patreon_0.12mm_PLA_Elegoo.Neptune.2D_1d9h40m.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13464532/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mmorning, 2023-11-25:
As I was typing this up, I started a new print and a random circle is on the print but not the sliced preview. 
Attached is a photo of the print while it is printing and the latest sliced profile. 

[Flexi Factory Bambu lab Panda_Patreon_0.12mm_PLA_Elegoo Neptune 2D_1d9h38m.zip](https://github.com/SoftFever/OrcaSlicer/files/13464551/Flexi.Factory.Bambu.lab.Panda_Patreon_0.12mm_PLA_Elegoo.Neptune.2D_1d9h38m.zip)
[image]

mmorning, 2023-11-26:
I reprinted this file again and I turned off Arc Fitting and the circles did not appear. I attached the sliced file again for comparison. Hopefully it helps. 

[Flexi Factory Bambu lab Panda_Patreon_0.12mm_PLA_Elegoo Neptune 2D_1d9h43m.zip](https://github.com/SoftFever/OrcaSlicer/files/13466522/Flexi.Factory.Bambu.lab.Panda_Patreon_0.12mm_PLA_Elegoo.Neptune.2D_1d9h43m.zip)

discip, 2023-11-28:
If you are running marlin you need to activate G2/G3 support.

Noisyfox, 2023-11-28:
This might be indeed a problem with arc fitting, which the printer went the opposite direction for the arc (or the slicer emitted the wrong gcode).

closed 2024-03-06 by github-actions[bot]