OrcaSlicer/OrcaSlicer #2885: Bridging not working?

Arakon, 2023-11-25:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11

### Additional system information

Irrelevant

### Printer

VZBot custom

### How to reproduce

- Load the attached project and slice
- Check Layer 11 and notice that the sacrificial layer in that position is not using bridging, but rather overhangs that turn around a corner, making them unprintable (at least cleanly)
- I generally have a feeling that a lot of places where bridges should be used, Orca implements overhangs instead.

### Actual results

Sacrificial layer in mid-air is printed as overhang instead of bridge
[image]
[image]

### Expected results

Bridging should be used. Example from Superslicer:
[image]

### Project file & Debug log uploads

[Y_gantry_bottom_left.zip](https://github.com/SoftFever/OrcaSlicer/files/13464592/Y_gantry_bottom_left.zip)
[debug_Fri_Nov_24_20_08_51_3192.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/13464593/debug_Fri_Nov_24_20_08_51_3192.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2023-11-26:
Disable the extra perimeters on overhang setting. 

[image]

Arakon, 2023-11-26:
Indeed, thanks.

closed 2023-11-26 by Arakon