OrcaSlicer/OrcaSlicer #2888: Crashing when I click pressure advance test

Brett19E7, 2023-11-25:
### OrcaSlicer Version

V1.8.0-beta1

### OS version

Windows 10

### Printer

Anycubic kobra neo 

### How to reproduce

Calibration
Click on pressure advance settings DDE method PA Tower or PA Pattern error message picture taken then click ok and shuts down 
[image]
[image]

### Actual results

[image]

### Expected results

Should slice the test so I can print it 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Downquark7, 2023-11-25:
Try updating to the latest version (1.8.0 official release)

Brett19E7, 2023-11-25:
Yeah I found that confusing witch was the correct one 
[image]

Downquark7, 2023-11-25:
Scroll down to assets of that box

Brett19E7, 2023-11-25:
Thank you I can see the windows installer  1.8.0 and the portable zip

Downquark7, 2023-11-25:
Installer is easier to use

Brett19E7, 2023-11-25:
Thank you just want it to work so I can run these tests lol appreciate it

Brett19E7, 2023-11-25:
[image]

Same issue

Noisyfox, 2023-11-28:
Could you post the crash log?

Brett19E7, 2023-11-28:
How do I find the crash log?

Noisyfox, 2023-11-28:
In the slicer's help menu there is an option called "open config folder" which you will find the logs there.

Brett19E7, 2023-11-28:
Thank you I will do that asap

Brett19E7, 2023-11-28:
There my only options I have

Brett19E7, 2023-11-29:
> I think you meant to post a screenshot, but it didn't work if you reply
> with email.

I sent a picture when I click help there is not the option for the file log
you told me about

Noisyfox, 2023-11-29:
Sorry but I don't understand what you are talking about. Stop replying via email if you want to post any pictures because that doesn't work.

Brett19E7, 2023-11-29:
[image]

Brett19E7, 2023-11-29:
Crant see the option for crash log

Noisyfox, 2023-11-29:
"Show configuration folder"

Brett19E7, 2023-11-29:
[image]

I believe it's the one where it creates a closure

Noisyfox, 2023-11-29:
Could you upload the log file?

Noisyfox, 2023-11-30:
Will the app crash if you select a build-in printer/filament/process profile?

silence48, 2023-12-31:
Another c0000005 ACCESS_VIOLATION
Related issues include #2921 #3375 #3340 #3292 #3157 #2896 #2766

closed 2024-04-07 by github-actions[bot]