OrcaSlicer/OrcaSlicer #2889: UI refuses to enlarge beyond a single monitor

tlhintoq, 2023-11-25:
### OrcaSlicer Version

1.7

### OS version

Win10x64

### Additional system information

i9 gen 11
See attached specs

### Printer

Any

### How to reproduce

On a dual monitor PC, try to resize to take advantage of all of your screen real estate.

### Actual results

Orca slicer refuses to resize taller than one monitor.

### Expected results

Orca slicer shouldn't be forcing the user to a smaller space during a drag.  There's already widgets in the title bar if someone want to snap to the window max.
[image]

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2023-11-28:
Anti-bot auto-close nudge remark

tlhintoq, 2023-12-03:
Hey bot - Don't close me now

tlhintoq, 2023-12-06:
Keep alive comment

foreachthing, 2023-12-07:
Looks like the OrcaSlicer windows' max. size is set to the startup monitor size.
Same issue on Windows 10/11.

foreachthing, 2023-12-07:
@tlhintoq I'm, by all means, no programmer 😆 .
I just wanted to let you and others know that the same for Windows is happening.

Noisyfox, 2023-12-08:
I'll take a look soon.

Noisyfox, 2023-12-09:
I think that's a limitation of Windows, it simply does not allow you to resize a window larger than the size of your largest monitor size.

foreachthing, 2023-12-09:
> Does any of the other software you are using that allows you to do so?

Almost every other software can do that (didn't try everything I have installed, but a whole bunch). So, it's not a Windows problem. It's the application not allowing to be larger than a single monitor.
Need more info, please let me know.

tlhintoq, 2023-12-09:
> Does any of the other software you are using that allows you to do so?

Absolutely - nearly any program should do it so long as there hasn't been some specific action taken by a program to deliberately stop it.
Its a pretty common case - Aren't you running Visual Studio?  It does it.  Pretty common to run code on a vertical monitor or on two monitors acting as one.
[image]
[image]
[image]
[image]

tlhintoq, 2023-12-14:
Keep alive

Noisyfox, 2023-12-16:
I've found the cause of this issue. Will raise a PR soon.

Noisyfox, 2023-12-16:
Could you give the pr build a try and see if it solves the problem? @tlhintoq

Noisyfox, 2023-12-16:
There are a few bugs found. Repoening this.

tlhintoq, 2023-12-16:
Watching for updates.
Sidenote: I haven't run Visual Studio in a couple years.  I don't even have it installed on my new laptop.  So pulling and compiling won't happen right away for me.  _{But I'm realizing I probably need to get back into this side of things}_

Noisyfox, 2023-12-17:
> Sidenote: I haven't run Visual Studio in a couple years. I don't even have it installed on my new laptop. So pulling and compiling won't happen right away for me.

You don't have to do this. We got CI build for every PRs, so you could just download the binary from it.

foreachthing, 2023-12-18:
@Noisyfox, which one should I try? https://github.com/SoftFever/OrcaSlicer/actions
Thanks!!

Noisyfox, 2023-12-18:
https://github.com/SoftFever/OrcaSlicer/actions/runs/7235901104?pr=3149
this one

foreachthing, 2023-12-18:
This PR works for me 😁 
This screenshot spans over 2x 4k monitors. It's an **AWESOME** sight! 👍🏼 

[image]

closed 2023-12-20 by SoftFever