OrcaSlicer/OrcaSlicer #2890: 1.8.0 flow inconsistencies, retraction disabled forcing rolling back to 1.7.0

R4toLoko, 2023-11-26:
### OrcaSlicer Version

1.8.0

### OS version

Any

### Additional system information

Any ( different folks are experiencing this problem )

### Printer

Any ( folks with different printers are experiencing this problem )

### How to reproduce

Any print, slice and usual, print it and you will notice blob everywhere, lack of retraction, poor printing quality.

### Actual results

This is a prototype test, top is 1.8.0. Notice how there is no retraction at all.
Bottom is 1.7.0, day and night difference.

[image]

-----

These are examples from Orca Discord, hopefully the users will provide more input:

Left is 1.8.0 while center and right are 1.8.0

[image]

-----

Retraction inconsistencies

[image]

Same user back to 1.7.0

[image]

### Expected results

To have retraction and flow working as they should.
It is worth to mention that 1.8.0 when first released had a flow bug requiring 1.8.0 RC release to fix it so something appears to still be off

### Project file & Debug log uploads

This happens with any print, not one print in specific.
Just slice and print.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

HakunMatat4, 2023-11-26:
It is fair to mentioned that the latest 1.8.0 version I have is **1.8.0-beta_nightly_rc2_2722.AppImage**
I noticed that **1.8.0 official** was released last week, I do not know if it has the same bug or it if the same version others with the same problem are running.
After spending days trying to figure out why the quality dropped, I am staying away from 1.8.0 for now.

igiannakas, 2023-11-26:
Any chance you can download the latest version and repeat the temp tower test? I’ve just done one with a new filament I was trying to calibrate with the latest release and it was fine. Also please post your project file so we can see the gcode output, it would help immensely.

HakunMatat4, 2023-11-26:
> Any chance you can download the latest version and repeat the temp tower test? I’ve just done one with a new filament I was trying to calibrate with the latest release and it was fine. Also please post your project file so we can see the gcode output, it would help immensely.

I am actually doing that now, Orca is not setting the shell pattern, I am setting it as Monolitc Line but it prints as Monolitic. 
But if I open a ticket about it, 1.7.0 will be the blame so I just started the print to check both quality overall and why Orca is doing its own thing. Will report asap.

HakunMatat4, 2023-11-26:
@igiannakas , 1.8.0 official appears to be normal, it "appears" but this is the only model I printed and too early to say that the problem was only happening with 1.8.0 RC2
At least the seam/retraction isn't looking a Frankenstein haha

[image]

[image]

I am doing another 3h print with this same filament, ASA, will keep monitoring things.
Now, what version are the folks with the same problem??
1.8.0?? 1.8.0 RC2??

Gonna have to wait for their input.

igiannakas, 2023-11-26:
There was an issue with the retraction that was fixed in RC2 https://github.com/SoftFever/OrcaSlicer/issues/2697 but I personally hadn’t noticed it as it only happened when wipe was off.

HakunMatat4, 2023-11-29:
1.8.0 official appears to indeed be printing fine without flow/calibration issue, and the problem was related to 1.8.0rc2 which myself and other folks were still using.

closed 2023-11-29 by HakunMatat4