OrcaSlicer/OrcaSlicer #2891: Fuzzy skin deviates way too much from expected geometry

andrewboktor, 2023-11-26:
### OrcaSlicer Version

1.7.0

### OS version

Archlinux

### Additional system information

Not relevant

### Printer

Not relevant

### How to reproduce

Enable Fuzzy skin Contour with 
1 (point distance) and 0.2 (thickness)
Use some geometry that has relatively sharp corners.

### Actual results

[image]

### Expected results

[image]
This screenshot is taken from a patched version of OrcaSlicer. I looked at the fuzzy skin code and compared to the Cura implementation and I think I identified and fixed the issue. Would love to submit a PR but I am no slicer expert + this would be my first PR in Orca, so some help would be greatly appreciated.

### Project file & Debug log uploads

[bugreport.zip](https://github.com/SoftFever/OrcaSlicer/files/13466261/bugreport.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-11-26:
> ### OrcaSlicer Version
> 1.7.0
> 
> ### OS version
> Archlinux
> 
> ### Additional system information
> Not relevant
> 
> ### Printer
> Not relevant
> 
> ### How to reproduce
> Enable Fuzzy skin Contour with 1 (point distance) and 0.2 (thickness) Use some geometry that has relatively sharp corners.
> 
> ### Actual results
> ![before](https://private-user-images.githubusercontent.com/1869452/285638253-5c16eda1-7321-4f2e-946f-f81e9b1e7879.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwMDUyOTQsIm5iZiI6MTcwMTAwNDk5NCwicGF0aCI6Ii8xODY5NDUyLzI4NTYzODI1My01YzE2ZWRhMS03MzIxLTRmMmUtOTQ2Zi1mODFlOWIxZTc4NzkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjZUMTMyMzE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzE5Y2E4YzlmZGQ1YjkzNjlmMWYxNzg3Y2JjMmNjNmUzYjEyZDFlNDJjODAwMTc0YjhkYTdhZjdlZmZkODFjNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.XYKUgrggQMj6t_kYwfnpe5ouHYOM83rTQov9mYR_5hI)
> 
> ### Expected results
> ![after](https://private-user-images.githubusercontent.com/1869452/285638259-ff1631b1-aa7b-4900-a75a-726a81a0e625.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwMDUyOTQsIm5iZiI6MTcwMTAwNDk5NCwicGF0aCI6Ii8xODY5NDUyLzI4NTYzODI1OS1mZjE2MzFiMS1hYTdiLTQ5MDAtYTc1YS03MjZhODFhMGU2MjUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjZUMTMyMzE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWZmMmU3NDI0MTc2OWEzZGMzNzUzZTRlNzBmNWNlYmIwOWI2MWU4NjIyNzY4OTM3ZTNhOTM1ZDA1YjExMGY0NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Ldt76U0zYpDtTRI-VS3GoOszLatR7RZQ6CBULKObWUQ) This screenshot is taken from a patched version of OrcaSlicer. I looked at the fuzzy skin code and compared to the Cura implementation and I think I identified and fixed the issue. Would love to submit a PR but I am no slicer expert + this would be my first PR in Orca, so some help would be greatly appreciated.
> 
> ### Project file & Debug log uploads
> [bugreport.zip](https://github.com/SoftFever/OrcaSlicer/files/13466261/bugreport.zip)
> 
> ### Checklist of files to include
> * [ ]  Log file
> * [ ]  Project file

Why dont you upload the changed files as a zip file to take a look here.

andrewboktor, 2023-11-26:
@igiannakas: even better, here is a patch
```patch
--- /home/andrew/git/aur/orca-slicer/src/OrcaSlicer/src/libslic3r/PerimeterGenerator.cpp
+++ /home/andrew/git/aur/orca-slicer-modified/src/OrcaSlicer/src/libslic3r/PerimeterGenerator.cpp
@@ -63,14 +63,15 @@
         double p0p1_size = p0p1.norm();
         // so that p0p1_size - dist_last_point evaulates to dist_left_over - p0p1_size
         double dist_last_point = dist_left_over + p0p1_size * 2.;
-        for (double p0pa_dist = dist_left_over; p0pa_dist < p0p1_size;
+        double p0pa_dist=dist_left_over;
+        for (; p0pa_dist < p0p1_size;
             p0pa_dist += min_dist_between_points + double(rand()) * range_random_point_dist / double(RAND_MAX))
         {
             double r = double(rand()) * (fuzzy_skin_thickness * 2.) / double(RAND_MAX) - fuzzy_skin_thickness;
             out.emplace_back(*p0 + (p0p1 * (p0pa_dist / p0p1_size) + perp(p0p1).cast<double>().normalized() * r).cast<coord_t>());
             dist_last_point = p0pa_dist;
         }
-        dist_left_over = p0p1_size - dist_last_point;
+        dist_left_over = p0pa_dist - p0p1_size;
         p0 = &p1;
     }
     while (out.size() < 3) {
@@ -105,12 +106,13 @@
         double p0p1_size = p0p1.norm();
         // so that p0p1_size - dist_last_point evaulates to dist_left_over - p0p1_size
         double dist_last_point = dist_left_over + p0p1_size * 2.;
-        for (double p0pa_dist = dist_left_over; p0pa_dist < p0p1_size; p0pa_dist += min_dist_between_points + double(rand()) * range_random_point_dist / double(RAND_MAX)) {
+        double p0pa_dist = dist_left_over;
+        for (; p0pa_dist < p0p1_size; p0pa_dist += min_dist_between_points + double(rand()) * range_random_point_dist / double(RAND_MAX)) {
             double r = double(rand()) * (fuzzy_skin_thickness * 2.) / double(RAND_MAX) - fuzzy_skin_thickness;
             out.emplace_back(p0->p + (p0p1 * (p0pa_dist / p0p1_size) + perp(p0p1).cast<double>().normalized() * r).cast<coord_t>(), p1.w, p1.perimeter_index);
             dist_last_point = p0pa_dist;
         }
-        dist_left_over = p0p1_size - dist_last_point;
+        dist_left_over = p0pa_dist - p0p1_size;
         p0 = &p1;
     }
 
@@ -1979,8 +1981,8 @@
                     current_position = best_path->junctions.back().p; //Pick the other end from where we started.
             }
         }
-
-        if (this->layer_id > 0 && this->config->fuzzy_skin != FuzzySkinType::None) {
+	// TODO: May be this should be a "Apply to first layer" setting
+        if (this->config->fuzzy_skin != FuzzySkinType::None) {
             std::vector<PerimeterGeneratorArachneExtrusion*> closed_loop_extrusions;
             for (PerimeterGeneratorArachneExtrusion& extrusion : ordered_extrusions)
                 if (extrusion.extrusion->inset_idx == 0) {
```

andrewboktor, 2023-11-27:
I just spun up a PR, it's probably easier that way.

PR #2903 (2023-11-28): Fuzzy skin fixes + on/off for first layer
Fixing fuzzy skin so it doesn't exit the desired geometry at line ends
Adding option to enable/disable fuzzy skin on first layer

Fixes #2891
closed 2023-11-28 by Noisyfox