OrcaSlicer/OrcaSlicer #2892: Surface pattern is being ignored

HakunMatat4, 2023-11-26:
### OrcaSlicer Version

1.8.0 official

### OS version

Linux Mint Cinnamon

### Printer

Voron Trident

### How to reproduce

- Set bottom or top surface to Monotonic Lines
- Print it to have Monotonic pattern instead

This happens with any print.

### Actual results

This is Monotonic Line pattern

[image]

Well, this isn't Monotonic Line or I am seeing things haha

[image]

Monotonic instead of Monotonic Line
[image]

### Expected results

Errrrrr, to have Monotonic Line instead of Monotonic would be nice haha

### Project file & Debug log uploads

[Orca_debug.log](https://github.com/SoftFever/OrcaSlicer/files/13466294/Orca_debug.log)
[[a]_rear_half_g2.stl.zip](https://github.com/SoftFever/OrcaSlicer/files/13466297/a._rear_half_g2.stl.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2023-11-26:
Looks fine to me

Top:
[image]

Bottom:
[image]

Infill:
[image]

Infill 2:
[image]

Project file:
[[a]_rear_half_g2.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13466856/a._rear_half_g2.3mf.zip)

Noisyfox, 2023-11-26:
Yeah I noticed that the monotonic pattern doesn't really work as intended, the extrusion direction is still changing. The only difference between monotonic and monotonic lines is the lines are connected or not.

HakunMatat4, 2023-11-26:
> Looks fine to me
> 

The slicer looks fine indeed but it prints Monotonic instead of Monotonic Line.

Those prints above, Orca totally ignored the surface pattern like it was never there.

So it either I'm doing something very wrong of this is one more Linux bug like the gcode file Orca could never load on Linux for releases after releases, and yet worked fine everywhere else.

I've checked the option over and over, all my printer have Monotonic pattern.

igiannakas, 2023-11-26:
Could you post your project file pls? Or even better the produced gcode file? Can check it in a gcode viewer.

Noisyfox, 2023-11-26:
The only difference between monotonic & monotonic line is whether the lines are connected. I guess this could again be a retracting problem or something similar that affect the disconnecting between lines.

igiannakas, 2023-11-26:
[image]
[image]

I’m trying to see it in my models but I can’t. Please post the project file and gcode output to investigate when you can.

igiannakas, 2023-11-26:
Ok may be going somewhere

Set Travel Distance threshold to 0 -> this should cause retractions to happen at every travel move
[image]

However retractions are not happening:
[image]

igiannakas, 2023-11-26:
This may cause the extruded line to appear like monotonic as there is ooze during the travel move before starting to print the next line.

igiannakas, 2023-11-26:
[image]

It does appear to retract when there is a travel move to another area but not when moving to the next line.

Noisyfox, 2023-11-26:
Have you compared with PS? I tried some test model on PS and it also doesn't retract at every lines.

igiannakas, 2023-11-26:
> Have you compared with PS? I tried some test model on PS and it also doesn't retract at every lines.

Yeap - PS does retract on every move. Orca used to as well as I remember using that to try and get better surface quality but was unnecessary in the end.

[image]

igiannakas, 2023-11-26:
Same issue (?) is also present on Bambu Studio:
[image]

HakunMatat4, 2023-11-26:
@igiannakas my bad, I thought I did that but sent the STL file instead, no comments.

[monotonic.zip](https://github.com/SoftFever/OrcaSlicer/files/13468174/monotonic.zip)

HakunMatat4, 2023-11-26:
Sorry, haven't had coffee yet, you also asked for the gcode file.

[[a]_rear_half_g2_0.2mm_ASA_TIME_11g.gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/13468196/a._rear_half_g2_0.2mm_ASA_TIME_11g.gcode.zip)

HakunMatat4, 2023-11-26:
Orca will show you that everything is right but something is not correct and the printer is printing Monotonic instead.

HakunMatat4, 2023-11-26:
Could that be the reason??
But again, you checked and retraction does no happen, perhaps left over from RC2.
You should have both the project and the gcode now.
Thanks a lot guys.

[image]

igiannakas, 2023-11-27:
> Sorry, haven't had coffee yet, you also asked for the gcode file.
> 
> [[a]_rear_half_g2_0.2mm_ASA_TIME_11g.gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/13468196/a._rear_half_g2_0.2mm_ASA_TIME_11g.gcode.zip)

So Gcode is fine - it shows lines instead of monotonic infill - see below:
[image]

[image]

Seems like a different issue to me. Probably you're getting a bit of ooze at the end of a move that spills over during the small travel move to start the next line and that causes it to appear like monotonic. 

Retraction shouldn't be necessary for this type of pattern, so the issue identified shouldn't be the cause for what you see. Retracting at every line in monotonic lines infill will extend your print by a considerable amount, especially if you also have z hop enabled, so in general it is not recommended to retract at the end of every line for this type of infill.

However this can be **potentially** caused by incorrect PA settings as the extruder doesn't release the pressure fast enough when slowing down near the end of the line causing it to print less crisp ends. This is just speculation on my side 

Finally you will see infill like the below:

[image]

This is from the detect narrow infill algorithm. It converts small infill movements to concentric and / or monotonic in small regions to speed up printing. You can disable that feature if you want under the strength tab, but again this should not be necessary as it will significantly extend the print time for little/no benefit as the areas are hidden by the top surface that is always monotonic lines.

The fix on retraction when ending a monotonic line infill with travel distance threshold set to 0 is still needed; more from a consistency point of view rather than a usefulness point of view, but if I were you I'd look more into PA and Square corner velocity as your specific causes for this.

HakunMatat4, 2023-11-27:
@igiannakas there's no way on Earth this is oozing. You can clearly see a perfect Monotonic pattern used instead of Monotonic Line. 
There no attempt to retract at all even so OrcaSlicer's legend shows it happening.
Oozing is very fine flow like a string and wouldn't be this uniform.

Indeed, the slicer also shows Line the the printed model shows otherwise.
Unless Klipper has a new feature setting pattern on its own. 

I'll try to print a test print with a different pattern like a cylinder with eccentric pattern.

 [image]

igiannakas, 2023-11-27:
It is indeed strange - however the Gcode doesnt show anything like this which makes it even more strange. Maybe worth trying clearing your print cache on Klipper and all stored files on Klipper just in case it's using a cached g-code file when you're printing.

HakunMatat4, 2023-11-29:
@igiannakas , I have printed a few things and boy, it just keeps getting weirder.
Some prints does show the Monotonic Line patter while others shows it as Monotonic.
Even a cube a printed with Monotonic now shows the top as Monotonic Line.
Now at least, I can see some prints with Line pattern.

I cannot narrow down wtf is going on lmao and yes I changed all 3 patterns, bottom, top and infill.
I am closing this ticket until I can easily replicate the problem.

closed 2023-11-29 by HakunMatat4