OrcaSlicer/OrcaSlicer #2893: M600 entry at the beginning of the print

bronsn, 2023-11-26:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11

### Printer

Ender 7 / K1

### How to reproduce

Setup a print File, add a second or third Filament, add at a specific layer height the Filament Change and slice the Objekt.
If i check the sliced gcode, there are 3 M600 Entry´s in the Code, 2 are correct but every time the Slicer add the M600 at the beginning of the Code so the printer stop with M600 after the Start Code

I have to manaally delete the M600 Entry in the Gcode.

### Actual results

M600 at the beginning of the print

### Expected results

M600 only th this point i added it in the slicer

### Project file & Debug log uploads

?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

staal54a, 2023-12-01:
I also seeing this issue

Eldenroot, 2023-12-02:
Confirmed

Krustyr47, 2023-12-05:
I see this as well.  M600 at the beginning.  However, also on every layer change not just the layer I selected.

samiresa, 2023-12-07:
Ender 3V2 can confirm it is happening for me as well, removing the fist m600 via notepad modification.

dral3x, 2023-12-08:
Same issue.

SwineFlew931, 2023-12-13:
I have been manually deleting the first M600. Sometimes i forget (it's that problem of being human). But this is so annoying that I am looking at learning Python just to hopefully write a post processing script to remove it. It seems a check box for an option to include/exclude that first change would solve the issue and give people with AMS/MMU units as well as manual change folks the power to configure it as they need.

dral3x, 2023-12-23:
I think it's related to the "Manual Filament Change" option, introduced in #2379.

If your printer (like my Sovol SV06) support just one filament, you have to turn ON that option in your printer profile.
After that, the M600 is not added anymore at the beginning of each print with multiple colors, but only on the layers in which you set the color switch.

borisb74, 2023-12-23:
> I think it's related to the "Manual Filament Change" option, introduced in #2379.
>
> If your printer (like my Sovol SV06) support just one filament, you have to turn ON that option in your printer [...]

Can confirm. After enabling the option for "Manual Filament Change" under Machine -> Multimaterial, the initial M600 is not in anymore. OrcaSlicer is version 1.8.1.

SwineFlew931, 2023-12-23:
> Can confirm. After enabling the option for "Manual Filament Change" under Machine -> Multimaterial, the initial M600 is not in anymore. OrcaSlicer is version 1.8.1.

It was there the whole time *face->palm*.

Thank you @dral3x  for pointing that out.

DBT85, 2024-01-03:
@dral3x thank you so much for that. I've only just started doing multi colours on my Ender 3 but having to strip the extra M600 after prime line out was a pain. Hopefully anyone else suffering the same issue will find this answer too!

manferlo81, 2024-02-17:
Thank you @dral3x, @DBT85, @borisb74 , @SwineFlew931. I have the "Manual Filament Change" on but it still happening to me. I will attach a test project I put together really quick for a different issue but as it turns out it has an M600 at the beginning.

[M600 String Issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14318079/M600.String.Issue.zip)

Eldenroot, 2024-02-29:
Prusa slicer maybe fixed this in the 2.7.2 version?

I am not sure. Anyway this is not good for nonAMS users.

@SoftFever please could you reply if it is somehow fixable?

samiresa, 2024-03-16:
[image]
This is how I fixed my issue, I think it is a bug, but this is a work around.

vgdh, 2024-03-17:
It's not a bug. 
If you change filament automatically you do need M600 at start of your Gcode because if there is no such command you will print with the last used filament. 
So when you choose **manual filament change** slicer doesn't emit M600 at the start because it is your job to put right filament in the machine.

closed 2024-06-24 by github-actions[bot]