OrcaSlicer/OrcaSlicer #2895: Max exceeds maximum extrusion - just in orca

d9C89, 2023-11-26:
Hello, I have a problem with the orcaslicer!

. I slice a part and always at exactly the same place it stops and spits out this error message. I only have that with the Orcaslicer. If I print with similar parameters with prusa, it's not a problem. So what do I have to set there?

[image]

mjfsch, 2023-11-27:
do you have any colour changes in this? i had the same issue with filament ramming - had to change max_extrude_cross_section in printer.cfg to allow for it.
If you have changed nozzle size then it may be related to that.

d9C89, 2023-11-27:
> do you have any colour changes in this? i had the same issue with filament ramming - had to change max_extrude_cross_section in printer.cfg to allow for it. If you have changed nozzle size then it may be related to that.

no, i have an idex, use 2 filaments. sv04. the print starts, an always on the same spot(after ca. 1h) error

StarkRG, 2023-11-29:
This sounds like the same bug as #1459