OrcaSlicer/OrcaSlicer #2896: Orca crashes when hitting the play button for video

tdwilli1, 2023-11-26:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11

### Additional system information

12th Gen Intel(R) Core(TM) i9-12900, 2400 Mhz, 16 Core(s), 24 Logical Processor(s)
Installed Physical Memory (RAM)	64.0 GB	
Intel(R) Arc(TM) A770   Graphics Driver Version	31.0.101.4953	

### Printer

Bambu Lab X1C

### How to reproduce

When selecting the Device View and hitting the Play button the application will crash and close after 3-5 seconds.

### Actual results

No screenshot available as the application quits.

### Expected results

The video from the printer should be playing. Project file makes no difference. Will crash with no file loaded.

### Project file & Debug log uploads

[crash_Sat_Nov_25_15_58_03_0.zip](https://github.com/SoftFever/OrcaSlicer/files/13467044/crash_Sat_Nov_25_15_58_03_0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2023-11-27:
Looks like an issue in Intel Arc graphic driver...

tdwilli1, 2023-11-27:
It would run the same driver. I have a NUC and it uses the same driver.

silence48, 2023-12-31:
Another c0000005 ACCESS_VIOLATION
Related issues include #2921 #3375 #3340 #3292 #3157

closed 2024-04-07 by github-actions[bot]