OrcaSlicer/OrcaSlicer #2897: Outlines are not closed - Gaps between the sliced part

Rockevzel, 2023-11-26:
I just added Files to the Slicer and sliced it. 
I cant find the issue. See picture below. The Lines should be closed

I didn't start the print yet. Maybe it's just a display error? 

[image]

igiannakas, 2023-11-27:
could you post the project file please?

DVSVIDEO, 2023-12-01:
This is not a forum, but a bug reporting section.
If you yourself don't know whether it's a mistake or not, why are you adding to the number of mistakes?
This is something for the developers to look over and generates unnecessary work.
What you can do:
Print it and find out if it is a display error or not.
You check your settings, because maybe the seam settings are out of order.
You try the same project with an earlier version.
After that, you open an error ticket after careful consideration if it is really the fault of the slicer.

closed 2024-03-09 by github-actions[bot]