OrcaSlicer/OrcaSlicer #2900: No Flush between Colorchange when Print "By object"

Uncail, 2023-11-26:
### OrcaSlicer Version

1.8.0

### OS version

Win 11

### Printer

P1P

### How to reproduce

I try to Print Color Samples with P1P and AMS. 4 Pieces in printorder By Object for Colorchange 

On 1.8.0 there are no Flush when i print "By Object"

On 1.8.0 RC2 it works fine

### Actual results

This is the Flushpart GCode from [OrcaSlicer V1.8.0 Official Release](https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.8.0)

; FLUSH_START
M400
M109 S230
G1 E2 F374 ;Compensate for filament spillage during waiting temperature
; FLUSH_END

and this is the Flushpart GCode from [OrcaSlicer V1.8.0 Release Candidate 2](https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.8.0-rc2)

; FLUSH_START
; always use highest temperature to flush
M400
M109 S250

G1 E23.7 F374 ; do not need pulsatile flushing for start part
G1 E0.719207 F50
G1 E8.27089 F374
G1 E0.719207 F50
G1 E8.27089 F374
G1 E0.719207 F50
G1 E8.27089 F374
G1 E0.719207 F50
G1 E8.27089 F374

; FLUSH_END

### Expected results

Above is the Result with 1.8.0 after a Colorchange.

Below is 1.8.0 RC2 after a Colorchange

[image]

### Project file & Debug log uploads

[Redline_PLA_Oxidrot_220_60.zip](https://github.com/SoftFever/OrcaSlicer/files/13467992/Redline_PLA_Oxidrot_220_60.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

16ruppt, 2023-11-27:
This seems to be a color change issue in general. No flushing between color change.

[image]

discip, 2023-11-28:
### Same issue here!
Was not sure why  nobody complained yet.

@SoftFever
@Noisyfox
Please have a look as this one is really critical.
I need to modify every single file by hand afterward to make this work as intended. 😭

@Uncail
This is not related to per object exclusively.
Please change the description to something like: **`No Flush between Colorchange`**. 😊

thanks in advance
kind regards

discip, 2023-11-28:
### Same issue here!
Was not sure why  nobody complained yet.

@SoftFever
@Noisyfox
@igiannakas
Please have a look as this one is really critical.
I need to modify every single file by hand afterward to make this work as intended. 😭

@Uncail
This is not related to per object exclusively.
Please change the description to something like: **`No Flush between Colorchange`**. 😊

thanks in advance
kind regards

igiannakas, 2023-11-28:
Haven’t noticed this and I did a multi color print just yesterday. Any chance you can upload the print gcode?

discip, 2023-11-28:
## Thank you for the almost instant reaction! 😃
> Any chance you can upload the print gcode and project file?

Yes, please see here: https://github.com/SoftFever/OrcaSlicer/pull/2819#issuecomment-1827978731

igiannakas, 2023-11-28:
> ## Thank you for the almost instant reaction! 😃
> > Any chance you can upload the print gcode and project file?
> 
> Yes, please see here: [#2819 (comment)](https://github.com/SoftFever/OrcaSlicer/pull/2819#issuecomment-1827978731)

I've pushed a fix here: https://github.com/SoftFever/OrcaSlicer/pull/2912

This is "severe" enough that it should warrant a patch release as purging is broken on BBL printers. @Noisyfox  @SoftFever

closed 2023-11-28 by Noisyfox

zviratko, 2023-11-28:
> ### Same issue here!
> Was not sure why nobody complained yet.
> 
> @SoftFever @Noisyfox @igiannakas Please have a look as this one is really critical. I need to modify every single file by hand afterwards to make this work as originally intended. 😭
> 
> @Uncail This is not related to per object exclusively. Please change the description to something like: **`No Flush between Colorchange`**. 😊
> 
> thanks in advance kind regards

Actually... https://github.com/SoftFever/OrcaSlicer/issues/2787

igiannakas, 2023-11-28:
Think this PR should have solved this too.

mvandenabeele, 2024-07-29:
I don't think this bug is solved even though the PR is merged. I can't get flushing going with the latest version 2.1.1, with or without prime tower. The volumes in the settings window always reset to 0 too, so I may be seeing another bug. It's very strange that this has gone unnoticed for so long. Makes me think I'm doing something wrong, but I can't figure out what that'd be. Printing on a Bambu X1C if it matters.

Schmudus, 2025-03-18:
Same here.
Also not being able to get purge volume overhanded to Klipper.