OrcaSlicer/OrcaSlicer #2902: Compiling on a Raspberry Pi 5 ends in windows drawing distorted horizontal lines

XenonFlits, 2023-11-26:
### OrcaSlicer Version

1.8.0

### OS version

Debian GNU/Linux 12 (bookworm) 64 bits

### Additional system information

Raspberry Pi 5 Model B Rev 1.0 - 4 GB
---
libgtk-3-0:arm64  1:3.24.38-2~deb12u1+rpt1+rpi1 - arm64 
libgtk2.0-0:arm64 2.24.33-2+rpt1 - arm64 
---
  *-graphics
       product: vc4drmfb
       physical id: 9
       logical name: /dev/fb0
       capabilities: fb
       configuration: depth=16 resolution=1920,1080

### Printer

None

### How to reproduce

sudo ./BuildLinux.sh -u
./BuildLinux.sh -dsr

### Actual results

Stipes in Windows 
[image]
and lots of GTK related complains in terminal starting the application.

### Expected results

Working application

### Project file & Debug log uploads

[2023-11-26 17:41:23.939164] [0x00007ffee7610040] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/dennis/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/dennis/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:83302): GLib-GObject-WARNING **: 17:41:25.960: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:41:25.960: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:83302): GLib-GObject-WARNING **: 17:41:25.962: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:41:25.962: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:83302): GLib-GObject-WARNING **: 17:41:25.965: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:41:25.965: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:83302): GLib-GObject-WARNING **: 17:41:25.969: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:41:25.969: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:83302): GLib-GObject-WARNING **: 17:41:25.972: invalid cast from 'wxPizza' to 'GtkCellLayout'

[... 314 lines cut ...]

(orca-slicer:83302): GLib-GObject-WARNING **: 17:42:22.700: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:42:22.701: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:83302): GLib-GObject-WARNING **: 17:42:22.701: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:42:22.701: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:83302): GLib-GObject-WARNING **: 17:42:22.708: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:42:22.708: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:83302): GLib-GObject-WARNING **: 17:42:22.710: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:42:22.710: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:83302): GLib-GObject-WARNING **: 17:42:22.711: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:83302): Gtk-CRITICAL **: 17:42:22.711: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

### Checklist of files to include

- [X] Log file
- [X] Project file

Kyle0122, 2023-11-27:
I have the same issue. It seems that the webview on the raspberry pi is not working. The home, Device and Project tab are broken but the prepare and preview tab works fine. I also sliced a 3d Benchy successfully.

SoftFever, 2023-11-30:
> Fantastic, after many hours of frustration, recompiling and searching the Internet this is the golden bullet !
> 
> After opening a terminal window, entering: << export WEBKIT_DISABLE_COMPOSITING_MODE=1 >> (Force Accelerated Compositing mode to be always off) and then run the app: All windows render OK!
> 
> But is this a feature or a bug? ;) Probably one of the Pi 5 acceleration libraries needs an update?
> 
> I will be testing the days ahead, but i guess : **This is the solution**. Thank you very much!
> 
> Still the console complaints a lot!
> 
> * malloc(): unsorted double linked list corrupted
> * (orca-slicer:4618): Gtk-CRITICAL **: 20:50:49.843: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> * (orca-slicer:4618): GLib-GObject-WARNING **: 20:50:49.844: invalid cast from 'wxPizza' to 'GtkCellLayout'
> * ** (orca-slicer:4618): CRITICAL **: 20:50:45.093: Cannot register URI scheme memory more than once
> * ** (orca-slicer:4618): CRITICAL **: 20:50:45.093: Cannot register URI scheme wxfs more than once
> * (orca-slicer:4618): Gtk-CRITICAL **: 20:50:39.808: gtk_widget_set_size_request: assertion 'width >= -1' failed
> * (orca-slicer:4618): Gtk-CRITICAL **: 20:50:39.207: gtk_window_resize: assertion 'height > 0' failed
> * (orca-slicer:4618): GLib-GObject-WARNING **: 20:50:39.118: invalid cast from 'wxPizza' to 'GtkCellLayout'
> 
> There are instantly about 175 lines with complaints during initialization with mostly repeatedly the lines above.

Just curious, how long does it take to build Orca on Rpi 5?

Kyle0122, 2023-12-02:
> Just curious, how long does it take to build Orca on Rpi 5?

I just built it again, it took about 100 minutes on a 4gb pi 5 with 4gb swap (using -j2, not including time to build deps).

The `export WEBKIT_DISABLE_COMPOSITING_MODE=1` is not working for me. The app opens, the webview renders correctly, but it crashes after about 12 seconds.

```
pi@raspberrypi:~/Projects/OrcaSlicer $ ./build/package/orca-slicer --debug 4
[2023-12-02 22:27:11.059408] [0x00007fff85170040] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/pi/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/pi/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot ope n shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:30004): GLib-GObject-WARNING **: 22:27:11.378: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30004): Gtk-CRITICAL **: 22:27:11.378: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
...............................................................
(orca-slicer:30004): Gtk-CRITICAL **: 22:27:11.398: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Could not determine the accessibility bus address
...............................................................
(orca-slicer:30004): GLib-GObject-WARNING **: 22:27:23.846: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30004): Gtk-CRITICAL **: 22:27:23.846: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Segmentation fault
```

I experimented with forcing gtk2, not forcing gtk2, with or without Appimage -- all resulted in crashes after 12 seconds.
Not knowing what to do, I randomly clicked on tabs and menus, and then I discovered that each time the app crashed at the moment when the mainsail webpage completed the `Initializing` loading animation.

Deleting the printer IP address stops the crashes, but then I cannot upload the gcode directly. I hope there's a compile option to remove the device tab. I'm not even using it on my laptop and removing it can help save a bit of memory on the Pi and other low-memory devices.

Ocraftyone, 2023-12-10:
I would like to add that the console also complains a lot on standard Linux too(in WSL2 and VM) mostly gtk

Skaifer, 2023-12-19:
Just hopped in to mention my comment [right there](https://github.com/SoftFever/OrcaSlicer/issues/2703#issuecomment-1861874669), maybe some of you would find it helpful in case if doing `export WEBKIT_DISABLE_COMPOSITING_MODE=1` command doesn't do the job.

Kyle0122, 2023-12-21:
A few days ago I tested the slicer under X11. Surprisingly, the `'WEBKIT_DISABLE_COMPOSITING_MODE=1'` workaround rendered the mainsail web page without any problem. It only crashes under wayland with mainsail os webpage.
So the problem is likely related to wayland and should only appear under Debian bookworm, since raspberry pi defaults to wayfire/wayland in this release.

Botspot, 2024-02-16:
Interested in adding OrcaSlicer to [our Raspberry Pi app store.](https://github.com/Botspot/pi-apps) Does anyone have a working build hosted anywhere?

Botspot, 2024-05-17:
Anyone?

closed 2024-08-23 by github-actions[bot]