OrcaSlicer/OrcaSlicer #2907: [FR] Option to allow slowdown for two external perimeters instead of just one.

DorjeDorf, 2023-11-27:
**Is your feature request related to a problem? Please describe.**
Not a problem, but one of the bigger differences I see between Orca and Kisslicer in terms of outside quality of the prints. 
There are surely a lot of algorithm differences, but even on simplest objects, Kiss will produce slightly better quality walls on the same settings as Orca. The main difference I see there is that for Kiss - two external perimeters are slowed down instead of just the most external one. 

**Which printers will be beneficial to this feature**
All of them. 

**Describe the solution you'd like**
An option in quality settings that would enable slowdown for two external perimiters instead of just one. 

**Describe alternatives you've considered**
Using Kiss, which is an option, but as much as I like it and the way it organizes settings (the best in all slicers imho), development on it is very flaky and delayed. It's missing a lot of features that are now common place on slicers like Orca. 

**Additional context**
Nothing to add.

igiannakas, 2023-11-28:
Enable the option to slow down for curled overhangs. It slows down multiple internal perimeters if needed and only where needed. 

[image]

DorjeDorf, 2023-11-28:
That's not what I want. I want all external perimeters at the whole length of them to be slower. 
The way kiss does it, let's say you have 5 perimeters. 
Perimeters 1-3 fron the inside go at full speed. Perimeter 4 goes slower than the other 3, but still a bit faster than the most external one (I believe it's halfway) and the final external one goes at the preset speed.

closed 2024-03-06 by github-actions[bot]