OrcaSlicer/OrcaSlicer #2915: Generic PETG will change to ASA - Bambulabs X1 & AMS

TechDude23, 2023-11-28:
### OrcaSlicer Version

1.8.0-beta 2

### OS version

Window 10

### Printer

Bambulabs X1

### How to reproduce

- Device tab
- Select Filament in the slot = generic PETG & confirm
- Next screen will change to ASA and it will not let you print with petg filament
- But if you switch it to Bambu labs basic petg it will work correctly and now you can print with petg
- This bug is also not in Bambu Studio Slicer 1.8.0.62 it's working correctly
- I don't know if the calibration tab and flow testing triggered this error ? 

[image]

[image]

### Actual results

Generic PETG will change to ASA filament on AMS icon

### Expected results

Generic PETG should show PETG on AMS filament icon

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

svenzman, 2023-11-29:
Happens here too, but it only bugged in AMS Slot 4 and External Spool Holder (P1S with AMS)

I reinstalled ervything and deleted data manually, not fixed.

TechDude23, 2023-12-13:
Version 1.9.0 DEV has been working correctly :)

closed 2023-12-13 by TechDude23