OrcaSlicer/OrcaSlicer #2921: Crashing/Closing While Slicing Complex or A lot of Layers

Zechs2205, 2023-11-29:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11

### Additional system information

[image]

### Printer

Bambu Lab  P1P

### How to reproduce

1. Open or Import a big file or complex file.
2. Slice using 0.12 layer high(or small layer in general) try to get 500 or more layers.
3. When it reaches 15%-80% crashes. 

### Actual results

Orca slicer just crashes/closes

### Expected results

Normal Slicing

### Project file & Debug log uploads

[debug_network_Wed_Nov_29_04_46_31.log](https://github.com/SoftFever/OrcaSlicer/files/13496223/debug_network_Wed_Nov_29_04_46_31.log)
[crash_Sat_Nov_25_01_55_07_0.log](https://github.com/SoftFever/OrcaSlicer/files/13496225/crash_Sat_Nov_25_01_55_07_0.log)
[crash_Sat_Nov_25_01_55_26_0.log](https://github.com/SoftFever/OrcaSlicer/files/13496226/crash_Sat_Nov_25_01_55_26_0.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

silence48, 2023-12-28:
strangely i have this issue in bambu slicer and orca, but not nearly as often in orca slicer. I also wonder if it's something to do with the e-cores / p-cores on these cpus.  I have z790 and 14900k cpu.

Zechs2205, 2023-12-28:
@silence48 what ram do you have? I have had issues with both bambu and orca. I also have the 13900K(similar) and z790.

silence48, 2023-12-31:
> @silence48 what ram do you have? I have had issues with both bambu and orca. I also have the 13900K(similar) and z790.

TeamGroup T-Create Expert 48GB Kit (2 x 24GB) DDR5-7200 PC5-57600 CL34 Dual Channel Desktop Memory Kit CTCWD548G7200HC34ADC01  It's not overclocked because the pc didn't run stable at 7200 mhz.  I am running the ram at 5600 mt/s

This issue has a possible relationship to [3279](https://github.com/SoftFever/OrcaSlicer/issues/3279)
[image]
[image]
[image]
[image]
[image]

Screen Recording of failure.
[image]
A relevant log:
```
Exceptions:

Exception Code :c0000005 ACCESS_VIOLATION

Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0xf 
Context :00000075E00F2A80 
ContextFlag : 0x10005f, EFlags: 0x10246 
Fault address:  0xCEF15BAE 0x1:0x24BAE C:\WINDOWS\SYSTEM32\ntdll.dll

Show CallStack:
```

Another log:
```log
Exceptions:

Exception Code :c0000005 ACCESS_VIOLATION

Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0xffffffff 
Context :00000036909F4840 
ContextFlag : 0x10005f, EFlags: 0x10206 
Fault address:  0x28FEB87C 0x1:0x1BEA87C B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll

Show CallStack:
7ffd28feb87c:hid_write [Unknown file][-1]
7ffd277fa1a2:cereal::detail::StaticObject<cereal::detail::bind_to_archives<Slic3r::DynamicPrintConfig,cereal::detail::`anonymous namespace'::polymorphic_binding_tag> >::instantiate [Unknown file][-1]
7ffd277fe3e7:cereal::detail::StaticObject<cereal::detail::bind_to_archives<Slic3r::DynamicPrintConfig,cereal::detail::`anonymous namespace'::polymorphic_binding_tag> >::instantiate [Unknown file][-1]
7ffd277fd92b:cereal::detail::StaticObject<cereal::detail::bind_to_archives<Slic3r::DynamicPrintConfig,cereal::detail::`anonymous namespace'::polymorphic_binding_tag> >::instantiate [Unknown file][-1]
7ffd2780b531:cereal::detail::StaticObject<cereal::detail::bind_to_archives<Slic3r::DynamicPrintConfig,cereal::detail::`anonymous namespace'::polymorphic_binding_tag> >::instantiate [Unknown file][-1]
7ffd27b5a214:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd27b58799:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd27b58ea3:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd27b590fe:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd27a09cd2:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd27a0bf23:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd27969a67:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd2798ed22:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd27493a25:Unknown [Unknown file][-1]
7ffd2748f232:Unknown [Unknown file][-1]
7ffd2749a17c:Unknown [Unknown file][-1]
7ffd274a18e9:Unknown [Unknown file][-1]
7ffd274a17ab:Unknown [Unknown file][-1]
7ffdccd71bb2:configthreadlocale [Unknown file][-1]
7ffdcd297344:BaseThreadInitThunk [Unknown file][-1]
7ffdcef426b1:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x28FEB87C 0x1:0x1BEA87C B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x277FA1A2 0x1:0x3F91A2 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x277FE3E7 0x1:0x3FD3E7 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x277FD92B 0x1:0x3FC92B B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x2780B531 0x1:0x40A531 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x27B5A214 0x1:0x759214 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x27B58799 0x1:0x757799 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x27B58EA3 0x1:0x757EA3 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x27B590FE 0x1:0x7580FE B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x27A09CD2 0x1:0x608CD2 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x27A0BF23 0x1:0x60AF23 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x27969A67 0x1:0x568A67 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x2798ED22 0x1:0x58DD22 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x27493A25 0x1:0x92A25 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x2748F232 0x1:0x8E232 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x2749A17C 0x1:0x9917C B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x274A18E9 0x1:0xA08E9 B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0x274A17AB 0x1:0xA07AB B:\OrcaSlicer_Windows_V1.8.1_portable\OrcaSlicer.dll
0xCCD71BB2 0x1:0x20BB2 C:\WINDOWS\System32\ucrtbase.dll
0xCD297344 0x1:0x16344 C:\WINDOWS\System32\KERNEL32.DLL
0xCEF426B1 0x1:0x516B1 C:\WINDOWS\SYSTEM32\ntdll.dll
```
TLDR;

It could be memory related, it could be related to threading and sometimes there's issues where stuff needs to be numa aware to properly utilize e-cores, but i'm thinking that's unrelated due to the fact i parked the e-cores and it still crashed.
The crashes happen at different times and seemingly different reasons (though i don't have examples of other reasons besides the logs i provided)
I am going to try to build the master branch on my local system instead of using the release.
I'm going to try running a system file scan and reinstalling the dll that the error comes from. 
I'll report back if i make any progress.
The same issue occurs in bambu slicer too, in fact in bambu it happens sooner.

silence48, 2023-12-31:
There is a likely relationship to [3375](https://github.com/SoftFever/OrcaSlicer/issues/3375)

silence48, 2023-12-31:
Another relationship to [3157](https://github.com/SoftFever/OrcaSlicer/issues/3157)

silence48, 2023-12-31:
[image]

Good news at least for me, enabling compatibility mode partially solved my problem, and i'm now able to slice most files. (though I still did see 2 crashes)

Zechs2205, 2024-01-05:
[image]

I upgraded my PC to DDR5 6800 and a new Motherboard. Still same issue. Same issue even after disabling e cores. 

@silence48 You're solution has solved it on my end. I wonder what the issue stems from.

chuckie1992, 2024-01-11:
@silence48 You are right, atleast on my end all crashes has stopped soon as i change affinity in my task manager to only the P cores( 0-7 for 8 P core cpu). I don't think parking the cpu solves this. Parked cpu's can still be activated when needed, you either need to fully disable them or only set affinity to cores 0-7. I have not tried compatibility mode.

@Zechs2205 i don't now why that wouldn't work for, i have exactly same setup except for ram i run slightly slower and Gskills. Doing the above helped me getting reliable slices off. I really feel like this is an E core issue since i've had this issue with every slicer i've tried so far and they all crash soon as i hit slice button. Doing the above solved for all of the slicers i tried(Cura, Creality Print and OrcaSlicer)

Zechs2205, 2024-02-01:
I figured it out. It's not E core its Hyperthreading. I used project lasso to test different hypothesis. I found that hyperthreading is casing problems with many different applications. I don't know why still but it seems to be a problem on the LGA1700 bc my previous I9 10th didnt have any of those issues.

closed 2024-05-09 by github-actions[bot]