OrcaSlicer/OrcaSlicer #2927: camera works in Mainsail but not OrcaSlicer

avxmw, 2023-11-29:
### OrcaSlicer Version

1.8.0

### OS version

openSUSE Tumbleweed

### Printer

Creality K1

### How to reproduce

1. Click "device" tab in OS
2. Look for camera picture
3. do not see it

Camera is configured via moonraker.con as described here: https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Configure-Camera

### Actual results

I am not getting a preview picture of the webcam in my printer on the 'device' tab of latest OrcaSlicer. Mainsail shows up and is functional.

Mainsail in Firefox and Mobileraker both show the camera.

Screenshot Firefox vs Orca
[image]

### Expected results

Being able to see my printer.

### Project file & Debug log uploads

[debug_Wed_Nov_29_17_31_01_10806.log.0.txt](https://github.com/SoftFever/OrcaSlicer/files/13503284/debug_Wed_Nov_29_17_31_01_10806.log.0.txt)
[debug_network_Wed_Nov_29_17_31_01.log.txt](https://github.com/SoftFever/OrcaSlicer/files/13503288/debug_network_Wed_Nov_29_17_31_01.log.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

avxmw, 2023-11-29:
For some reason the cam works in OrcaSlicer if I set it up as http://ip:4409/webcam/?action={snapshot,stream} but not if setup as http://ip:8080/?action=stream while in Firefox both work and even in parallel.

crazymind1337, 2023-12-01:
Camera not working for me too. I am using OrcaSlicer 1.8.0 with Creality K1 Max with latest firmware from Creality. If I configure the suggestion from @avxmw camera only shows on firefox but just partly:
[image]

Nothing shows on OrcaSlicer and Chrome.

ServerNinja, 2023-12-14:
This seems to be a problem with moonraker. If you have your webcam configured in the moonraker.cfg, it will NOT work in OrcaSlicer. If you comment that section out and just use the mainsail config for the camera, it starts to work in OrcaSlicer. The problem is that I use the webcam configuration in moonraker.cfg in order to get the camera to work on other devices (I have another instance of klipperscreen I run on an Rpi on my desk so I can monitor prints while I'm working on other things).
[image]

padcom, 2023-12-29:
I am using [Docker](https://www.docker.com/) to run [Klipper](https://github.com/Klipper3d/klipper) using [mkuf/prind](https://github.com/mkuf/prind) composition. The default content of [moonraker.conf](https://github.com/mkuf/prind/blob/main/config/moonraker.conf) does not contain `[webcam]` configuration yet still I have the same problem. 

Any idea what might be causing it? I'd very much prefer the full experience using just OrcaSlicer than to have a separate tab open with mainsail to just view the stream.

padcom, 2023-12-29:
It seems that in my setup there is no `moonraker-obico` container running, so having or not the `[webcam ...]` section in `moonraker.cfg` makes little difference.

padcom, 2023-12-30:
I have kind of made it work with the Adaptive MJPEG-Streamer:

[image]

My motivation to change to the Adaptive service wasn't even because of the webcam - it's just the default 15 fps saturates my WiFi network too much. Maybe this will help others get it to work too.

avxmw, 2023-12-31:
@padcom Indeed, changing moonraker.conf to have `mjpegstreamer-adaptive` also gives me a picture in OrcaSlicer.

Thank you for this, works as a workaround. However I would not call this a fix as at the least this should be documented.

padcom, 2023-12-31:
It's definitely not a "fix". It's a lousy workaround at best. That does not change the fact that OrcaSlicer with Mainsail embedded looks like the space shuttle control center at the Kennedy Space Center. The only thing that could make it look better is if Orca would support dark theme because now, to make it look the same as the rest of Orca, Mainsail needs to be switched to light theme...

[image]

closed 2024-04-07 by github-actions[bot]

bwees, 2024-11-17:
I am still having this issue, even when using MJPEG-adaptive. WebRTC works but my host is not powerful enough to run it and stutters. Is there some debug mode I can go into to see why the WebView isn't able to load it?