OrcaSlicer/OrcaSlicer #2928: Doesn't run in a remote desktop on a VM - OpenGL error

jbuszkie, 2023-11-29:
### OrcaSlicer Version

1.8.0

### OS version

win10

### Printer

none

### How to reproduce

Try to start

### Actual results

Program doesn't run.
I get the error "The application cannot run normally because your  OpenGL version is lower than 2.0"
Thent he program quits.
[image]

### Expected results

The program should run.
Bambu studio 1.7.7.89 runs fine.  Prusa slicer runs fine.

### Project file & Debug log uploads

Can't get configuration 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jbuszkie, 2023-11-29:
Same issue with V1.7.0

Noisyfox, 2023-11-30:
You can copy the "mesa" folder from BBS or PS into Orca and it should work. We will fix this in a future release.

closed 2023-12-01 by jbuszkie

Mechanikerlegion511, 2024-01-28:
Hello guys, in version **1.9.0** **WIN10x64** there was also a problem: I had to copy the ***mesa*** folder from PrusaSlicer to ***C:\Program Files\OrcaSlicer*.**
_but under Linux everything works fine on the same laptop._
**The problem is only with Windows 10.**

Antebios, 2025-03-13:
I copied the `mesa` from my SuperSlicer directory into OrcaSlicer and it worked!!!

InsanityOnABun, 2025-11-04:
> This ticket should be reopened so it can be properly resolved; I shouldn't have to installed a different slicer to fix a problem with Orca.

Agreed, this issue isn't fixed, there is just a workaround that requires an install of another slicer.