OrcaSlicer/OrcaSlicer #2932: Sign Windows Executables

Ocraftyone, 2023-11-30:
Add signing to Windows executable (both the installer and the executable in the portable version)

Running an executable that is not signed on Windows shows a warning that it could contain malware etc and could turn away some users.

SoftFever, 2023-12-04:
I thought about it.
But I feel it's not very necessary at the moment because of the hassles.
This blog summarizes it well: https://melatonin.dev/blog/how-to-code-sign-windows-installers-with-an-ev-cert-on-github-actions/

kylek29, 2023-12-04:
^^ That explains why I see so many indie and small company programs that aren't even bothering with code-signing for Windows now, as a user I've gotten used to by-passing the malware disclaimer, their current system is contradictory to the whole purpose of the malware warning. They badly need to simplify that.

Ocraftyone, 2023-12-10:
Yeah, that looks like a huge headache. Hopefully it becomes easier and more accessible soon.

closed 2023-12-10 by Ocraftyone