OrcaSlicer/OrcaSlicer #2933: Keyboard Shortcuts Not Working on MacOS (V1.8.0) in the Device tab

ajaytanna, 2023-11-30:
### OrcaSlicer Version

1.8.0

### OS version

MacOS Sonoma 14.1.1

### Additional system information

Apple M2 Pro
16 GB - 512 GB

### Printer

Creality Ender-3 S1 Pro

### How to reproduce

Install and check for keyboard shortcuts in the Device tab - Hide and Quit menu items do not work. In other tabs they do.

### Actual results

Keyboard shortcuts do not work

### Expected results

Keyboard shortcuts should work

### Project file & Debug log uploads

n the latest OrcaSlicer V1.8.0 Official Release on MacOS Arm64 on a mac Mini M2 Pro the application shortcuts Hide OrcaSlicer, Hide Others and Quit OrcaSlicer are not working when in the device tab. They are working okay when in the Prepare and Preview tabs. This is on MacOS Sonoma 14.1.1.  The same thing happens in the latest Nightly also.
[[a]_a_tensioner_pinned.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13507928/a._a_tensioner_pinned.3mf.zip)
[Log.zip](https://github.com/SoftFever/OrcaSlicer/files/13507929/Log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ajaytanna, 2023-12-02:
Update response to V1.8.1 Nightly
The shortcuts do not work as indicated above. In fact they don't work in the Home, Device and the Project tabs. But they do work in the Prepare and Preview tabs. Really disturbs the workflow. More so after enjoying the luxury of the Full Mainsail view in the Device tab.

ajaytanna, 2023-12-02:
Update response to V1.9.0Dev
Same behaviour as before. Left clicking on the blank menu area just to right of "Project" makes the keyboard shortcuts work.

nevack, 2024-03-06:
Still true as of 1.9.1

closed 2024-06-12 by github-actions[bot]