OrcaSlicer/OrcaSlicer #2937: UI issue on MacOS: scrolling panels

superfloh247, 2023-11-30:
### OrcaSlicer Version

e8a053e

### OS version

MacOS 14.1.1

### Additional system information

Apple Silicon 14.1.1

### Printer

Bambu P1P

### How to reproduce

scroll any of the panels, eg Quality

### Actual results

panel slides over other UI elements

### Expected results

panel should slide behind other UI elements

### Project file & Debug log uploads

[image]

now scroll down

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-11-30:
Yeah this only happens when you build locally. Seems to be a bug with newer Xcode. Tried to debug this a few months back but no luck.

Eldenroot, 2023-11-30:
There are many macos fixem in bambu studio 1.8.1 version (released today). We should sync all useful changes and fixes into Orca. 1.8.0 is not synced too

Noisyfox, 2023-12-01:
I've found the solution https://github.com/wxWidgets/wxWidgets/issues/23916
Will try backporting their fixes to Orca.

Noisyfox, 2023-12-19:
Should have been fixed now.

closed 2023-12-19 by Noisyfox

superfloh247, 2023-12-23:
aaaaaaaaaand back again in 374f78c

not fixed anymore

Noisyfox, 2023-12-23:
> aaaaaaaaaand back again in [374f78c](https://github.com/SoftFever/OrcaSlicer/commit/374f78c7689b375c9fe842a2d4b8807eab70f176)
> 
> not fixed anymore

Yeah @SoftFever is working on it right now.