OrcaSlicer/OrcaSlicer #2946: Bambu P1S slams against front and back on inital start - each print

insp1re1, 2023-11-30:
### OrcaSlicer Version

8

### OS version

win11

### Printer

bambu p1s

### How to reproduce

start any sliced print

### Actual results

Bambu P1S slams against front and back on inital start - each print

### Expected results

being gentle each start.

### Project file & Debug log uploads

happens with latest orca, doesnt happen with bambu studio latest beta.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mjfsch, 2023-12-01:
Really silly, but have you checked your set up to use the correct printer? ive seen a lot of reports of this sort of thing happening if people accidently have it set to the A1 mini instead of the P1 or X1

mjfsch, 2023-12-01:
in that case, probably a good idea to upload a project file and a sliced gcode file so someone can have a look and see if they can work out whats going on - outside of selecting the wrong printer profile ive not seen much mention of this happening, so the more info the better

poppinfresh69, 2023-12-09:
This is not a bug. Bambus don't have limit switches and instead sense the current change from the stepper motors locking up as they hit the physical limit to determine where home is.

insp1re1, 2023-12-09:
> This is not a bug. Bambus don't have limit switches and instead sense the current change from the stepper motors locking up as they hit the physical limit to determine where home is.

you must be confused, it was indeed an issue with a faulty sdcard causing gcode corruption, so I can close this now.

stepper motors currents made zero difference to what it did.

closed 2023-12-09 by insp1re1