OrcaSlicer/OrcaSlicer #2947: Make profiles more portable making Bed Texture and bed Model not use absolute path to user .

victornpb, 2023-12-01:
**Is your feature request related to a problem? Please describe.**
Currently the profile saves a path to where the models are on your computer, and Orca directly loads files from wherever they are in your computer. This is bad for 2 reasons, if you move a file it no longer works, and using the same profile on 2 different computers, when syncing them makes the visualization breaks, since files aren't on the same place.

**Which printers will be beneficial to this feature**
Every printer

**Describe the solution you'd like**
I like to propose to inline both files as base64 inside the JSON config files, this would make sharing profiles super convenient and more portable. And that guarantees that using import configs is the only step you have to do to have everything neat and working for a printer.

**Describe alternatives you've considered**
Another alternative is to export those files as well when using export configs with standard names like model.stl, texture.png and making so you can import everything as a zip and orca would keep those files inside App dir, instead random user's directories.

**Additional context**

Add any other context or screenshots about the feature request here.

Noisyfox, 2023-12-01:
> I like to propose to inline both files as base64 inside the JSON config files

That's an interesting idea.

TomLeenheer, 2023-12-01:
On the same note it would be nice to be able to overwrite the system printers like the genereral klipper printer definition alowing you to give it your own name and printbed and picture. So you can realy define your own custom printer.

Noisyfox, 2023-12-20:
> On the same note it would be nice to be able to overwrite the system printers like the genereral klipper printer definition alowing you to give it your own name and printbed and picture. So you can realy define your own custom printer.

This feature will be ported over from BambuStudio in the next release.

closed 2024-03-27 by github-actions[bot]