OrcaSlicer/OrcaSlicer #2956: shifted icon

discip, 2023-12-01:
### OrcaSlicer Version

[ae9a2d5](https://github.com/SoftFever/OrcaSlicer/suites/18675832753/artifacts/1086755360)

### OS version

Linux (Manjaro)

### Printer

unrelated

### How to reproduce

Printer > click symbol (edit preset)

### Actual results

Symbol under another symbol.

### Expected results

sgsegf

### Project file & Debug log uploads

drht

### Checklist of files to include

- [X] Log file
- [ ] Project file

Eldenroot, 2023-12-09:
You are right :(

Ocraftyone, 2023-12-10:
Noticed this too

discip, 2023-12-22:
@Noisyfox @igiannakas @SoftFever
Does anyone have any idea about this?

discip, 2023-12-22:
@Noisyfox @igiannakas @SoftFever
Does anyone have any idea about this?
This is still present in [0f2a223](https://github.com/SoftFever/OrcaSlicer/actions/runs/7300261688)

discip, 2023-12-22:
@SoftFever @Noisyfox @igiannakas
Does anyone have any idea about this?
This is still present in [0f2a223](https://github.com/SoftFever/OrcaSlicer/actions/runs/7300261688)

discip, 2023-12-22:
@SoftFever @Noisyfox @igiannakas
Does anyone have any idea about this?
This is still present in [0f2a223](https://github.com/SoftFever/OrcaSlicer/actions/runs/7300261688).

igiannakas, 2023-12-22:
Unfortunately I can't replicate it as I dont see these icons on the Mac (and dont have another computer :() Will defer to Noisyfox / Ocraftyone for this one

discip, 2023-12-23:
Still present even after [Revert to WxWidgets 3.1.5 (#3249)](https://github.com/SoftFever/OrcaSlicer/actions/runs/7307508790).

discip, 2024-01-27:
@SoftFever
@Noisyfox

Any idea yet?

discip, 2024-04-27:
Unfortunately this is still present using Manjaro Gnome.

discip, 2024-05-06:
@yw4z
Could you please have a look at this?

[image]

Additionally the icon is shifted very far to the left. Also in **`Printer settings`**:

[image]

I'm not sure if this also appears this way under Windows but using Linux (Manjaro Gnome) it certainly does.

Thanks in advance!

yw4z, 2024-05-06:
i coulnt replicate issue on my system (windows 10) but this line responsible from that icon and DISABLE_UNDO_SYS also appears on  OG_CustomCtrl.cpp

https://github.com/SoftFever/OrcaSlicer/blob/e40d7780d596ed5e61973fbffb452f5391355a13/src/slic3r/GUI/Tab.cpp#L385-L388

here is how that area should looks like 
[image]

@jamincollins can you look at this im not using linux :)

jamincollins, 2024-05-06:
I'll take a look

discip, 2024-07-24:
@jamincollins
Any progress so far? 😊

jamincollins, 2024-07-24:
No longer contributing to this project

discip, 2024-07-24:
Oh, that's a shame.
Take care. 😊

closed 2024-10-24 by discip