OrcaSlicer/OrcaSlicer #2960: Add Tool option

Tutankamon00, 2023-12-02:
Hi, i have ERCF MMU, and at start my extruder are without filament loaded, it need te Tool Command Tx(T0-T1-T2 ETC..) to load filament, my problem is OrcaSlicer put Tool command after my Start G-code(i use macro called AVVIO_STAMPA), please add option for add tool selection before start gcode, thank you
[image]

Ocraftyone, 2023-12-10:
@SoftFever can you think of any reason to not always set the tool before starting gcode or would that cause issues with some of the implementations

Ocraftyone, 2023-12-10:
Or, we could do something similar to how setting temperatures works where if it is set in the starting gcode, it isn't emmited automatically

SoftFever, 2023-12-10:
> Hi, i have ERCF MMU, and at start my extruder are without filament loaded, it need te Tool Command Tx(T0-T1-T2 ETC..) to load filament, my problem is OrcaSlicer put Tool command after my Start G-code(i use macro called AVVIO_STAMPA), please add option for add tool selection before start gcode, thank you ![immagine](https://private-user-images.githubusercontent.com/142754987/287406116-2245d0ae-07f6-4602-8be2-ff08ea62ce6a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxNzExNDQsIm5iZiI6MTcwMjE3MDg0NCwicGF0aCI6Ii8xNDI3NTQ5ODcvMjg3NDA2MTE2LTIyNDVkMGFlLTA3ZjYtNDYwMi04YmUyLWZmMDhlYTYyY2U2YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxMFQwMTE0MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02OWM0ZDg3M2I1ZTNjNDRkMjU4N2M4MTU4M2VhODQzYzg2ZjVlNjY0YTI0MWIwMDg0MGU5ZDQxY2NiODIwNjBiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.nvxRz1VI6m4w2ZcauLeM_GZiTmLtGgt_swbcvSh5fso)

You can insert `T[initial_tool]` in your custom g-code.
ERCF will automatically ignore the following T2 command if the T2 channel is already in use.

SoftFever, 2023-12-10:
> @SoftFever can you think of any reason to not always set the tool before starting gcode or would that cause issues with some of the implementations

Hi @Ocraftyone 

I followed PrusaSlicer's behaviors here.
My line of thought is to maintain consistency with PrusaSlicer's behavior unless there are compelling reasons to do otherwise. Various MMU devices and firmware and DIY users handle tool change logic differently, but they have all adapted to the logic used by PrusaSlicer.

Ocraftyone, 2023-12-10:
> You can insert `T[initial_tool]` in your custom g-code.
> ERCF will automatically ignore the following T2 command if the T2 channel is already in use.

Glad to see this is already implemented. It is much more dynamic than just hard coding the position. 

It should also be mentioned that you can use gcode variables to help set the initial tool dynamically.

Tutankamon00, 2023-12-11:
> T[initial_tool]

Thank you very much!!!

closed 2024-03-12 by Tutankamon00