OrcaSlicer/OrcaSlicer #2963: Add a new settings - clean nozzle after XXX layers

Eldenroot, 2023-12-02:
**Is your feature request related to a problem? Please describe.**
This new simple settings could help to keep the nozzle unclogged (for example with some materials - PLA-CF etc. from some brands). Purge is nice :) 

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
Add a new settings to clean the nozzle after XXX layers (XXX = any number, configurable)

**Describe alternatives you've considered**
---

**Additional context**
---

stephen3303, 2023-12-16:
I wish the Bambu Lab prints had a button to order a nozzle clean mid-print at my discretion.

Eldenroot, 2023-12-17:
Yes, that would be nice... maybe some day some "macros" will be available from Babulab side :) who knows.

Now we need this setting :)

Sortam, 2023-12-19:
I vote for having both options :)

Eldenroot, 2024-03-19:
B

vgdh, 2024-03-19:
If you use klipper it't pretty straight forward to implement as a macro.

foreachthing, 2024-03-19:
Easy as pie in OrcaSlicer, too! 😉   _(I don't use Klipper)_
Just add this to either before or after layer change GCode in Printer Settings (advanced mode):

```
{if (layer_num - int(layer_num / 25) * 25) == 0}   ; START CLEAN
G92 E0           ; reset extruder reference
G1 E-8           ; retract 8mm - yes, bowden tube
G92 E0           ; reset extruder reference
G91              ; relative coordinates
G0 Z1            ; move up a bit
G90              ; absolute positioning
;  G0 X0 Y210 F9000 ; UM2, move to far left corner   < ---- edit this line to tell the printer where to go.
G1 E23           ; EXTRUDE 15mm, CLEAN Nozzle (23, because we're in -8)
G1 E-8           : Retract once more
G91              ; relative coordinates
G0 Z-1           ; move back down
G90              ; absolute coordinates
G1 E8            ; unretract
G92 E0           ; reset extruder reference
; END CLEAN    {endif}
```

This will emit this code ever 25th layer.

[image]

**CAUTION** This code works on our Ultimaker2 and does probably not work on your printer!!! CHECK THE GCODE FIRST!
It's your fault if you damage your printer - you've been warned...

vgdh, 2024-03-19:
> {if (layer_num - int(layer_num / 25) * 25) == 0}   ; START CLEAN

WOW !!! I didn't know that you can implement a logic at the slicer side. So cool! Thank you.

Eldenroot, 2024-03-19:
Nice, could be incorporated into gui?

closed 2024-06-26 by github-actions[bot]