OrcaSlicer/OrcaSlicer #2967: Settings bug preventing Orca from running

Hirn74656, 2023-12-02:
### OrcaSlicer Version

1.8

### OS version

Windows 10

### Printer

Ender 7

### How to reproduce

I think it can be reproduced while going to the speed calibration and closing it without saving, but since i cannot open orca anymore I am unsure

### Actual results

Hi,

I was working in an instance setting up a speed test when my computer shut down, now every time I try to open Orca I get an error about vase mode and it will not open not matter which selection I choose (yes or no).

I have tried uninstalling and reinstalling Orca, transferring the crash logs and even deleting the .zip and trying over again but nothing works.

I really like the slicer and would like to continue to use it

### Expected results

I expected it to work

### Project file & Debug log uploads

[image]
[image]
[crash_Mon_Nov_27_15_16_02_0.log](https://github.com/SoftFever/OrcaSlicer/files/13536954/crash_Mon_Nov_27_15_16_02_0.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Zechs2205, 2023-12-04:
Same issue

Ocraftyone, 2023-12-10:
Try deleting your current settings. Do this by deleting %appdata%/OrcaSlicer. If you have any profiles you want to keep, backup this directory before deleting.

Zechs2205, 2023-12-10:
Does not work. I already tried it and I just tried it too.  I read on reddit that someone had the same problem. I think its relate to the CPU I'm using. Its a Intel I9 13900K. I tried:
- Deleting Software and reinstalling
- Deleting  %appdata% profile
- Reinstalling Windows. 
None has worked. It must be related to the CPU's allocation of memory.

closed 2024-03-18 by github-actions[bot]