OrcaSlicer/OrcaSlicer #2976: Hitting organic support with nozzle in travel movements

Santars2022, 2023-12-03:
### OrcaSlicer Version

All versions

### OS version

Windows 10

### Printer

Any printer

### How to reproduce

It depends on printed model

### Actual results

Screenshot in attachment

[image]

### Expected results

Add: setting for avoid support when travelling or add z-hop only on supports

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Santars2022, 2023-12-04:
Hello! This is well known problem, but nobody is fixing it. Only Cura slicer programmers had fixed this issue. From now on i'm on cura slicer intil issue is fixed.
If you search on google this problem, you will find a lot of acticles about prusa slicer, there is the same issue.
As you see on the picture above, organic support is hit by nozzle even when travelled from tree to tree over already printed tree.

Santars2022, 2023-12-04:
It's very sad that orca slicer programmers still didn't fix it

gaaat98, 2024-04-10:
I am affected too! Several small and detailed prints fail when slicing with Orca but not with Cura because the nozzle knocks off supports or other printed parts! Z-Hop is sometimes a solution but introduces a lot of stringing, while "avoid crossing walls" doesn't seem to be working and does not apply to supports...
Avoid printed parts and avoid supports when printing are basically the only options I miss from Cura... I would really like to see them implemented in Orca (and I guess many other users too)!

gaaat98, 2024-07-13:
#3254 presents a similar problem; path planning should be improved by introducing Cura settings such as combing and avoid printed parts/supports when travelling. I would really like to help but I don't know where to start...

closed 2024-10-20 by github-actions[bot]