OrcaSlicer/OrcaSlicer #2993: Retraction speed not considered in gcode - can cause extruder stop during print

rommulaner, 2023-12-04:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Printer

Anycubic Vyper, Chiron

### How to reproduce

- configure retraction an deretraction speed in printer configuration (in my case Marlin 2)
- retractions are made with print speed during wipe
- detretractions are done with the configured detretractions speed

### Actual results

- E-Stepper can loose steps
- Extruder can stop working during print
- Extruder makes unusual noises

### Expected results

Retraction move at wipe with configured speed for retraction

### Project file & Debug log uploads

Gcode example with retraction speed 45 and detretraction speed 35 on whipe tower test:

;TYPE:Outer wall
G1 F6000
G1 X111.635 Y128.579 E5.58807
G1 X111.203 Y129.154 E5.60942
G1 X110.639 Y129.6 E5.63077
G1 X109.98 Y129.887 E5.65211
G1 X109.27 Y129.997 E5.67345
G1 X108.555 Y129.923 E5.69479
G1 X107.882 Y129.669 E5.71615
G1 X107.297 Y129.251 E5.7375
G1 X106.837 Y128.699 E5.75883
G1 X106.533 Y128.047 E5.78019
G1 X106.405 Y127.34 E5.80152
G1 X106.466 Y126.603 E5.82348
G1 F3243.018
G1 X106.718 Y125.908 E5.84543
G1 F1200
G1 X107.111 Y125.349 E5.86572
G1 F1153.815
G1 X107.143 Y125.302 E5.86741
G1 F992.211
G1 X107.712 Y124.83 E5.88936
G1 F909.923
G1 X108.385 Y124.523 E5.91132
G1 F904.65
G1 X109.115 Y124.403 E5.93329
G1 F889.614
G1 X109.851 Y124.478 E5.95525
G1 X110.541 Y124.744 E5.97721
G1 F967.584
G1 X111.138 Y125.181 E5.99917
G1 F1120.316
G1 X111.248 Y125.32 E6.00444
M106 S255
G1 F1200
G1 X111.599 Y125.759 E6.02113
G1 F2498.964
G1 X111.892 Y126.438 E6.04308
**G1 F5659.883**
G1 X112.069 Y127.451 **E6.07362**
;WIPE_START
G1 X111.71 Y128.384 **E.97362**
;WIPE_END
G92 E0
G1 Z51.9 F7200
G1 X108.081 Y125.417 Z51.9
G1 Z51.5
**G1 E5.1 F2100**
M204 P2500
;TYPE:Internal solid infill
;WIDTH:0.38292
G1 F6000
G1 X108.591 Y125.199 E5.11568
G1 X109.105 Y125.116 E5.13041
G1 X109.686 Y125.169 E5.14689
G1 X110.169 Y125.343 E5.16141
; stop printing object retraction_tower.stl id:20482699039392 copy 0
;WIPE_START
G1 F6000
G1 X109.686 Y125.169 E2.54195
G1 X109.201 Y125.124 E.06141
;WIPE_END
G92 E0
M106 S0

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2023-12-07:
Wipe is considered as a normal extrusion action(extruding while moving the print head). So that's correct.

rommulaner, 2023-12-07:
Wouldn't it make more sense to calculate the retraction length at retraction speed for the wipe at print speed and retract the rest of the retraction with a simple G1 -### F(retraction_speed) after the wipe?
So only retract a small portion during the wipe to stay with the extruder speed at the set retraction speed and retract the rest after the wipe.

Otherwise you can't use the wipe function for bowden printers that have high retraction values of about 5mm, because it always runs into the speedlimits of the firmware or looses steps if they are set too high. My Vyper and my Chiron both stopped extruding while printing a pressure advanced calibration part because of this too high speeds.

igiannakas, 2023-12-10:
Potential fix published above. Please test and report back

igiannakas, 2023-12-12:
A new PR has been pushed above that implements this suggestion. Would appreciate some feedback

igiannakas, 2023-12-27:
This is now fixed in 1.9. Please either review and close or provide feedback if not working.