OrcaSlicer/OrcaSlicer #2997: Not Show Images(table/icons/printers)

XEGARE, 2023-12-04:
### OrcaSlicer Version

1.8.1 and older

### OS version

Windows 11 Pro Build 22631.2715

### Additional system information

CPU: AMD Ryzen 7 5800X
Memory: 64GB
Display Card: Nvidia RTX 2060 Super

Display: HDR 3440x1440

### Printer

Not selected

### How to reproduce

First Launch

### Actual results

[image]
[image]

### Expected results

Images must be visible

### Project file & Debug log uploads

FIX: The program breaks if there is a Cyrillic alphabet in the path of the executable files. Most likely it will break in some other languages.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-12-05:
Make sure the install path does not contain non-ascii characters.

XEGARE, 2023-12-05:
> Make sure the install path does not contain non-ascii characters.

Then the program works correctly.  I think the bug will be fixed in the future.  Thank you.

closed 2023-12-19 by SoftFever