OrcaSlicer/OrcaSlicer #3000: 1.9: GUI issue on Windows when display scaling is > 100%

Noisyfox, 2023-12-05:
### OrcaSlicer Version

1.9

### OS version

Windows 11

### Printer

N/A

### How to reproduce

1. Set display scaling to say 200%
2. Open the 1.9 dev build

### Actual results

1. Icon & text size issue: 
[image]
The icons are too large, while the title text is too small.

2. Right mouse menu y position is off. When you right click on the plate, the context menu is no longer positioned at the mouse cursor, but shifted up.

### Expected results

It should look the same as 1.8.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-12-05:
(At least) part of the issue was caused by not using the bitmapbundle, as now the size of the toolbar icons are correct after https://github.com/Noisyfox/OrcaSlicer/commit/0a0376a76d1d6b781e50e5a30d546df7dd241094 :

[image]

Noisyfox, 2023-12-05:
Also, I believe 
https://github.com/SoftFever/OrcaSlicer/blob/main/src/slic3r/GUI/wxExtensions.cpp#L892-L894
might be caused by the line here
https://github.com/SoftFever/OrcaSlicer/blob/25a055491ee8ef664497d480e1d0e06a71bbea59/src/slic3r/GUI/Widgets/SwitchButton.cpp#L141

Ocraftyone, 2023-12-06:
You are at the same position that I am. I just forgot to respond to the comment. There also seems to be an issue with the text size in the top bar not scaling correctly, but I do not see any code responsible for that. I am still looking. 

As for the switch button, I don't believe that should be causing the issue for the Global/Objects switcher. It is most likely the cause for the advanced toggle though. I think that to fix the advanced toggle I just need to re-implement the scale to size function that BBL implemented for regular bitmaps

Ocraftyone, 2023-12-06:
Got text scaling fixed. Not sure what changed that prevents it from not being done automatically, but oh well 🤷.

Ocraftyone, 2023-12-06:
yes. I noticed that too during development. Though I fixed it but it must have been reapplied when I went back through the patches

Ocraftyone, 2023-12-06:
@Noisyfox I think that I figured out the scaling issue. For some reason, when PS updated their code to use the new versions of wx, they rewrote how they get the svg data. Instead of parsing the SVG data and using the resulting image to help scale the image based on the target px, it just reads the raw svg file and passes that to wx to parse and load. In wx's implementation, you are required to give an exact dimension and we were passing the px value for both width and height. This is what was giving the funky images when the input was not exactly square. This should be pretty easy to remedy by switching back to the nsvg function that parses the read data into an nsvg image, get the scaled dimensions of the svg, then pass to wx to create the bitmap bundle. I am working on an implementation rn.

tsmith35, 2023-12-20:
Is there a fix I can make to current source to regain the normal sizing from before? I have to run compatibility mode to get the normal layout, but that comes with some quirky behavior and fuzzy graphics. Any help is appreciated. The new features look very nice.

Noisyfox, 2023-12-20:
> Is there a fix I can make to current source to regain the normal sizing from before? I have to run compatibility mode to get the normal layout, but that comes with some quirky behavior and fuzzy graphics. Any help is appreciated. The new features look very nice.

Try this PR build https://github.com/SoftFever/OrcaSlicer/pull/3032

Ocraftyone, 2023-12-20:
I just merged main into the PR tied to this issue. This will allow you to run 1.9dev with the current patches. You can get an artifact [here](https://github.com/SoftFever/OrcaSlicer/actions/runs/7269481813) when it finishes building.

tsmith35, 2023-12-20:
Okay, seems to look great and work great. The only remaining strangeness is on the calibration tab. Screenshots attached. Windows 10 22H2 with compatibility features disabled for OrcaSlicer. Resolution is 3840x2160 with 175% scaling.
[image]
[image]
[image]

tsmith35, 2023-12-20:
Oops, forgot this one...
[image]

closed 2023-12-21 by SoftFever