OrcaSlicer/OrcaSlicer #3002: 1.9: Performance issues

igiannakas, 2023-12-05:
### OrcaSlicer Version

1.9 Dev

### OS version

Mac OS Sonoma

### Additional system information

M1 Pro 16 gig

### Printer

Baby X1 carbon

### How to reproduce

Raising ticket so we can track this performance issue on 1.9 Dev.

1. 1.8.1 runs smooth as butter on the 3d Preview
2. 1.9 dev with #2934 reverted runs slower but workable. 
3. Then #2934 cause it to be completely unworkable for large models.

PR #2934 needs optimisation for large polygon models. In addition there is a further PR that I have not identified yet that is causing performance degradation.

### Actual results

N/A

### Expected results

N/A

### Project file & Debug log uploads

https://www.printables.com/model/331094-articulated-christmas-dragon

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2023-12-05:
Another candidate for the performance issue is this one: https://github.com/SoftFever/OrcaSlicer/pull/2603 Noticed performance degradation with it compared to before.

igiannakas, 2023-12-05:
@Noisyfox happy to help troubleshooting.

zviratko, 2023-12-05:
> Once complete, I'd appreciate some feedback. To run the build you need to run this: `xattr -dr com.apple.quarantine /Applications/OrcaSlicer.app`

Interesting, this fixes opening the build that claimed to be damaged before. I thought CTRL+rightclick+Open works around that as this message was new to me :-)

I can confirm that this build fixes it. Still not snappy, but definitely fixes this particular isue.
Thanks.

igiannakas, 2023-12-05:
Yeap you need to run that command for any build that is not produced from the main Softfever branch as they are not signed.

zviratko, 2023-12-05:
Yeah, but previously (pre-Sonoma?) it would say it's from an unknown developer, or that it can't be verified and could have been opened via ctrl+right+open. Now it says  "is damaged and can't be opened" which confused me.

Noisyfox, 2023-12-05:
It works smoothly on my old intel mbp, and I have no idea what Apple did with their m1 chips that caused this issue. I guess I'll have to add an option to disable the outline then.

igiannakas, 2023-12-05:
> It works smoothly on my old intel mbp, and I have no idea what Apple did with their m1 chips that caused this issue. I guess I'll have to add an option to disable the outline then.

Yeah there is something funny going on here. The outline is a big one but also https://github.com/SoftFever/OrcaSlicer/pull/2603 is causing performance issues. Taking a build before this one was merged is performing fine with high poly models, while after it’s stuttering significantly.

Prusa is not stuttering at all with the same file linked above.

igiannakas, 2023-12-05:
Im trying to get the performance monitor running to grab some stats out of Xcode - any pointers here to help you debug this would be greatly appreciated :)

Noisyfox, 2023-12-06:
Could you give the https://github.com/SoftFever/OrcaSlicer/pull/2819 build a try and see how well it performs? This PR includes some changes that should help.

igiannakas, 2023-12-06:
> Could you give the #2819 build a try and see how well it performs? This PR includes some changes that should help.

Will do - waiting for it to finish compiling :) 

Attached the perf monitors from the current 1.9 Dev branch (without the above 2819 push) just in case you see anything funny there.
[Profile - 1.9.trace.zip](https://github.com/SoftFever/OrcaSlicer/files/13583258/Profile.-.1.9.trace.zip)

igiannakas, 2023-12-06:
Perf traces above are with the dragon loaded in the first screen and moving the platter around. This action causes stuttering

zviratko, 2023-12-06:
> Could you give the #2819 build a try and see how well it performs? This PR includes some changes that should help.

Smooth sailing on my M1 Max even with the large dragon. Thanks!

igiannakas, 2023-12-06:
> Could you give the #2819 build a try and see how well it performs? This PR includes some changes that should help.

Absolutely perfect this one #2819 ! Butter smooth! Closing issue. Thank you!

closed 2023-12-06 by igiannakas