OrcaSlicer/OrcaSlicer #3012: [feature] - Backup/ Export ALL config files and presets

pixel-shock, 2023-12-05:
**Is your feature request related to a problem? Please describe.**
- As I am regulary exporting my configs as a backup into dropbox or whatever it would be nice to export **all** profiles, configs etc. and not just the selected/ active one. 

**Which printers will be beneficial to this feature**
- all

**Describe the solution you'd like**
- As a user I want to click on the "export configs" and I want OrcaSlicer to export all config files and profiles - like a full backup of all my created files.

**Describe alternatives you've considered**
- none

**Additional context**
- none

Best & thank you for reading this :wink:

yourbuddydinec, 2023-12-05:
Yes please. I just noticed today that it does this. I had been exporting for months only to today find out each was only the currently selected stuff. Backups are horrible like this if you have a ton of filaments or profiles

Ocraftyone, 2023-12-10:
This would only solve the backup problem, but you may be able to use the snapshots feature

closed 2024-03-16 by github-actions[bot]

KondratokGK, 2024-08-04:
YEEEEEEES PLEASE!

Furthen64, 2024-08-18:
I would love to have that feature <3

HazzzeHunter, 2025-02-02:
Any update here ?

Antebios, 2025-02-08:
Top search result to find out how to back up settings brought me here.  So, I don't think there is a solution?

yourbuddydinec, 2025-02-08:
> Top search result to find out how to back up settings brought me here. So, I don't think there is a solution?

I've been using this to backup for a while. Works great though I havent used it to restore in like 6 months. 

http://ilkos-home.blogspot.com/p/orca-slicer-pm.html

HazzzeHunter, 2025-03-05:
@yourbuddydinec  thanks works greate