OrcaSlicer/OrcaSlicer #3015: "Other layers" bed temperature not working after changing settings

leoshusar, 2023-12-05:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11

### Printer

Klipper Generic

### How to reproduce

1. Slice something (M140 is present)
2. Change e.g. brim settings
3. Slice (M140 is missing)
4. Revert the brim settings
5. Slice (M140 is still missing)
6. Change profile
7. Slice (M140 is present again)

### Actual results

See the video:

https://github.com/SoftFever/OrcaSlicer/assets/13891066/90334718-4f9c-4c2d-b5a2-e76e64c55218

### Expected results

M140 should be present every time

### Project file & Debug log uploads

[after-issue.zip](https://github.com/SoftFever/OrcaSlicer/files/13572825/after-issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

nishantsnair, 2023-12-13:
I notice the same on v1.8.0

btCRO, 2023-12-14:
I noticed if you change any speed setting in profile M140 is missing from gcode for layer 2. Reseting profile does not help but switching to another profile helps. I've tested a little on V1.8.1 and on some print settings change M140 is still generated but on some like print speed in 100% it is missing from generated gcode. This really sohud not happen.

btCRO, 2023-12-15:
And today I found that disabling automatic brim would forgot to set M140 at second layer in 100% of situation I'we tried.

cbaugher, 2023-12-24:
Confirmed.  Setting brim to disabled removes M140 from second layer setup.  OS v1.8.1.
Very weird.

mralien, 2024-01-15:
Same happens with 1.9.0.
My steps:
1) Set "Other layers" temp and sliced - M140 present in gcode
2) Changed filament cost (just to reproduce, it happens with other settings too) and sliced - M140 disappeared

To make it reappear I've changed temp to new value (different from old one), closed and re-sliced, then changed back to one I need. Changing temperature like X (required one) -> Y -> X without re-slicing at Y value doesn't bring back missing M140.

lhndo, 2024-04-07:
Confirmed, and seems to be brim related as @cbaugher said.

vgdh, 2024-04-07:
same https://github.com/SoftFever/OrcaSlicer/issues/4676

antonyburden, 2024-05-23:
I too have this issue on latest v.21dev

phreshxmetal, 2024-05-25:
I also have this issue, ver 2.0

JPR75, 2024-05-25:
Same problem with Orca 2.0.0 / Windows 11 / gcode flavor: Klipper.
I don't use brim. Most of the time the temperature change is done correctly but not always.

closed 2024-06-04 by SoftFever