OrcaSlicer/OrcaSlicer #3018: Random crash, then won't restart unless delete config files

gpeden, 2023-12-06:
### OrcaSlicer Version

1.8.1

### OS version

Linux Mint (Ubuntu)

### Additional system information

System:
  Kernel: 5.15.0-89-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 5.8.4
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.2 Victoria base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Micro-Star product: MS-7C56 v: 1.0 serial: <superuser required>
  Mobo: Micro-Star model: MPG B550 GAMING PLUS (MS-7C56) v: 1.0 serial: <superuser required>
    UEFI: American Megatrends LLC. v: 1.F0 date: 10/11/2023
Battery:
  Device-1: hidpp_battery_0 model: Logitech MX Keys Wireless Keyboard serial: <filter>
    charge: 55% (should be ignored) status: Discharging
CPU:
  Info: 8-core model: AMD Ryzen 7 5800X bits: 64 type: MT MCP arch: Zen 3 rev: 0 cache:
    L1: 512 KiB L2: 4 MiB L3: 32 MiB
  Speed (MHz): avg: 3363 high: 4697 min/max: 2200/3800 boost: enabled cores: 1: 2826 2: 2240
    3: 3880 4: 4296 5: 3952 6: 4667 7: 4697 8: 4633 9: 4612 10: 3690 11: 2240 12: 2214 13: 2200
    14: 2200 15: 2200 16: 3262 bogomips: 121597
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA GM206 [GeForce GTX 950] vendor: PNY driver: N/A pcie: speed: 2.5 GT/s lanes: 16
    bus-ID: 2b:00.0 chip-ID: 10de:1402
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nouveau,vesa
    unloaded: fbdev,modesetting gpu: N/A display-ID: :0 screens: 1
  Screen-1: 0 s-res: 2560x1440 s-dpi: 96
  Monitor-1: default res: 2560x1440 size: N/A
  OpenGL: renderer: llvmpipe (LLVM 15.0.7 256 bits) v: 4.5 Mesa 23.0.4-0ubuntu1~22.04.1
    direct render: Yes
Audio:
  Device-1: NVIDIA GM206 High Definition Audio vendor: PNY driver: snd_hda_intel v: kernel pcie:
    speed: 2.5 GT/s lanes: 16 bus-ID: 2b:00.1 chip-ID: 10de:0fba
  Device-2: AMD Starship/Matisse HD Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel
    pcie: speed: 16 GT/s lanes: 16 bus-ID: 2d:00.4 chip-ID: 1022:1487
  Sound Server-1: ALSA v: k5.15.0-89-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Micro-Star MSI
    driver: r8169 v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 2a:00.0
    chip-ID: 10ec:8168
  IF: enp42s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
Bluetooth:
  Device-1: Broadcom BCM20702A0 Bluetooth 4.0 type: USB driver: btusb v: 0.8 bus-ID: 1-6:5
    chip-ID: 0a5c:21e8
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.0
    sub-v: 220e
Drives:
  Local Storage: total: 1.95 TiB used: 535.12 GiB (26.8%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WDBRPG5000ANC-WRSN size: 465.76 GiB
    speed: 31.6 Gb/s lanes: 4 serial: <filter> temp: 39.9 C
  ID-2: /dev/sda vendor: Western Digital model: WD6400AAKS-22A7B2 size: 596.17 GiB
    speed: 3.0 Gb/s serial: <filter>
  ID-3: /dev/sdb vendor: Western Digital model: WD10EZEX-00BN5A0 size: 931.51 GiB
    speed: 6.0 Gb/s serial: <filter>
Partition:
  ID-1: / size: 365.24 GiB used: 34.91 GiB (9.6%) fs: ext4 dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 190.5 MiB used: 6.1 MiB (3.2%) fs: vfat dev: /dev/nvme0n1p1
  ID-3: /home size: 915.82 GiB used: 250.1 GiB (27.3%) fs: ext4 dev: /dev/sdb1
Swap:
  ID-1: swap-1 type: partition size: 93.13 GiB used: 0 KiB (0.0%) priority: -2 dev: /dev/nvme0n1p3
Sensors:
  Message: No sensor data found. Is lm-sensors configured?
Repos:
  Packages: 3355 apt: 3327 flatpak: 5 snap: 23
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/1password.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/1password-archive-keyring.gpg] https: //downloads.1password.com/linux/debian/amd64 stable main
  Active apt repos in: /etc/apt/sources.list.d/appimagelauncher-team-stable-jammy.list
    1: deb [signed-by=/etc/apt/keyrings/appimagelauncher-team-stable-jammy.gpg] https: //ppa.launchpadcontent.net/appimagelauncher-team/stable/ubuntu jammy main
  Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list
    1: deb [signed-by=/usr/share/keyrings/brave-browser-archive-keyring.gpg arch=amd64] https: //brave-browser-apt-release.s3.brave.com/ stable main
  Active apt repos in: /etc/apt/sources.list.d/github-cli.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/githubcli-archive-keyring.gpg] https: //cli.github.com/packages stable main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/nordvpn.list
    1: deb https: //repo.nordvpn.com/deb/nordvpn/debian stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com victoria main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  No active apt repos in: /etc/apt/sources.list.d/playonlinux.list
  No active apt repos in: /etc/apt/sources.list.d/steam-beta.list
  Active apt repos in: /etc/apt/sources.list.d/steam-stable.list
    1: deb [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
    2: deb-src [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
  Active apt repos in: /etc/apt/sources.list.d/yubico-stable-focal.list
    1: deb http: //ppa.launchpad.net/yubico/stable/ubuntu jammy main
Info:
  Processes: 454 Uptime: 2d 5h 42m Memory: 62.72 GiB used: 12.41 GiB (19.8%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12/9 Client: Cinnamon v: 5.8.4 inxi: 3.3.13

### Printer

Bambulab C1 Carbon

### How to reproduce

Run Orca Slicer.  Notice when crashes (random not sure how or how long).  

### Actual results

App won't restart.  I noticed if I delete the two .conf files in .config/OrcaSlicer it will start again (and all my configuration is lost).

### Expected results

Should not crash, but if it does should not leave in a state that it can't restart.

### Project file & Debug log uploads

[debug_network_Tue_Dec_05_19_47_07.log](https://github.com/SoftFever/OrcaSlicer/files/13575642/debug_network_Tue_Dec_05_19_47_07.log)
[debug_network_Tue_Dec_05_19_46_37.log](https://github.com/SoftFever/OrcaSlicer/files/13575643/debug_network_Tue_Dec_05_19_46_37.log)
[debug_network_Tue_Dec_05_18_45_25.log](https://github.com/SoftFever/OrcaSlicer/files/13575644/debug_network_Tue_Dec_05_18_45_25.log)
[debug_network_Tue_Dec_05_18_44_28.log](https://github.com/SoftFever/OrcaSlicer/files/13575645/debug_network_Tue_Dec_05_18_44_28.log)
[debug_network_Tue_Dec_05_10_37_02.log](https://github.com/SoftFever/OrcaSlicer/files/13575646/debug_network_Tue_Dec_05_10_37_02.log)
[debug_network_Tue_Dec_05_14_06_29.log](https://github.com/SoftFever/OrcaSlicer/files/13575647/debug_network_Tue_Dec_05_14_06_29.log)
[debug_network_Tue_Dec_05_10_35_24.log](https://github.com/SoftFever/OrcaSlicer/files/13575648/debug_network_Tue_Dec_05_10_35_24.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

gpeden, 2023-12-06:
sorry i don't have more to go on.

riktanius, 2023-12-06:
I am getting this error too

fdosani, 2023-12-16:
Same for me, but it seems to happen when updating any printer settings for my Ender 3 S1 Pro. Have to delete the `.config` folder for Orca to start working again.

SoftFever, 2023-12-31:
Hi all,

 can you attached the other log file(not this network log)

gpeden, 2023-12-31:
sorry i haven't had this crash in a while.  next time i do i'll upload all the logs.

closed 2024-04-07 by github-actions[bot]