OrcaSlicer/OrcaSlicer #3021: Login with Google crashes OrcaSlicer

dialtone, 2023-12-06:
### OrcaSlicer Version

1.8.1

### OS version

MacOS 14.1.2 (latest)

### Printer

Bambulab X1C

### How to reproduce

1. Launch OrcaSlicer
2. Install Network Plugin
3. Click on Login/Register
4. Click on Google Logo under Login with other accounts
5. Orca slicer crashes

### Actual results

Orca slicer crashes

### Expected results

Orca Slicer should log you in

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13587203/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

closed 2024-03-14 by github-actions[bot]

blake-tisbury, 2024-05-04:
Having the same exact issue with the P1S

whattheflukz, 2024-11-16:
also having this isse