OrcaSlicer/OrcaSlicer #3028: [MacOS] [AnyVersion] Filament configuration is removed on update and cannot be saved in future.

AKharytonchyk, 2023-12-07:
### OrcaSlicer Version

1.8.1

### OS version

macOS 14.1.2 (23B92)

### Additional system information

Chip: Apple M1 Pro
Memory: 16GB

### Printer

Ender 3 S1 Klipper

### How to reproduce

1. Install any version of OrcaSlicer for the first time.
2. Edit the filament configuration.
3. Save the configuration.
4. Download another version from the GitHub page.
5. Install the new version by running the .dmg file and replacing the current application.
6. Open OrcaSlicer and verify saved filament settings. Only the default settings will be available.
7. Update the settings as required.
8. Save the updated settings and close OrcaSlicer.
9. Reopen OrcaSlicer and verify the saved filament settings. It may be seen that no settings were saved previously.

### Actual results

Filament configuration is not saved between application updates.
The file configuration cannot be saved after an application update.

**The macOS update process caused OrcaSlicers to lose rights to the config folder, resulting in the issue.**

### Expected results

Filament configuration must be saved between application updates.
The file configuration should persist between application usages.

### Project file & Debug log uploads

[debug_Thu_Dec_07_09_03_12_52998.log](https://github.com/SoftFever/OrcaSlicer/files/13596369/debug_Thu_Dec_07_09_03_12_52998.log)
[debug_network_Thu_Dec_07_09_03_12.log](https://github.com/SoftFever/OrcaSlicer/files/13596370/debug_network_Thu_Dec_07_09_03_12.log)
[debug_Thu_Dec_07_09_02_09_52690.log](https://github.com/SoftFever/OrcaSlicer/files/13596371/debug_Thu_Dec_07_09_02_09_52690.log)
[debug_network_Thu_Dec_07_09_02_09.log](https://github.com/SoftFever/OrcaSlicer/files/13596372/debug_network_Thu_Dec_07_09_02_09.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

liftbag, 2023-12-09:
I'm sure this problem also happened to me some time ago. I'm still rebuilding the profiles, either starting from scratch, or copying the parameters from SuperSlicer. But I'm currently testing at least a couple of builds a day, without losing my profiles anymore.

AKharytonchyk, 2023-12-12:
@liftbag, do you have any suggestions for completely cleaning up the Orca installation and starting from scratch? Could there have been an issue with the initial setup version?

liftbag, 2023-12-16:
> @liftbag, do you have any suggestions for completely cleaning up the Orca installation and starting from scratch? Could there have been an issue with the initial setup version?

I believe that deleting the app, along with the folder ~/Library/Application Support/OrcaSlicer should remove any presets and allow for a clean install.

AKharytonchyk, 2023-12-19:
@liftbag, so I run this to clean everything up.
```bash
pkill -f OrcaSlicer

sudo rm -rf /Applications/OrcaSlicer.app

rm -rf ~/Library/Application\ Support/OrcaSlicer
rm -rf ~/Library/Preferences/com.orcaslicer.*
rm -rf ~/Library/Caches/com.orcaslicer.*       
rm -rf ~/Library/Saved\ Application\ State/com.orcaslicer.* 

rm -rf ~/Library/Logs/OrcaSlicer
rm -rf ~/Library/Cookies/com.orcaslicer.*.binarycookies 
```

I installed a fresh copy of an application from GitHub Releases. 
I went through the initial process of printer setup.
I logged into OrcaSlicer. It displayed a message that configs were saved in the cloud and asked whether I would like to load them. 
It loaded my Printer Configs, but the Filament and Process settings did not. Moreover, settings are still not persistent between application loads.

closed 2024-03-27 by github-actions[bot]