OrcaSlicer/OrcaSlicer #3036: Internal thick bridges overlap sparse infill

Downquark7, 2023-12-07:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Set Layer Height to 0.08mm
Turn on Thick internal bridges (on by default)
Slice object

### Actual results

[image]

The infill overlaps with the thick bridge causing surface defects

### Expected results

The internal infill should stop below the internal bridge instead of inside of it. Maybe the overlap doesn't cause issues at 0.2mm layer height but it causes problems on 0.08mm layer height.

### Project file & Debug log uploads

[3DBenchy.zip](https://github.com/SoftFever/OrcaSlicer/files/13604624/3DBenchy.zip)
3mf file inside of zip

### Checklist of files to include

- [ ] Log file
- [X] Project file

Downquark7, 2023-12-08:
Thick internal bridges enabled makes the bridges too thick (.4mm bridge height)
Thick internal bridges disabled makes the bridges fail (.08mm bridge height)
[image]
[3DBenchy side by side comparison.zip](https://github.com/SoftFever/OrcaSlicer/files/13607746/3DBenchy.side.by.side.comparison.zip)

If this is just a "I need to tune the internal bridge flow ratio" and not just a sub optimal method of calculating internal bridges please close this issue.

Noisyfox, 2023-12-09:
[image]
Have you tried changing this value?

Downquark7, 2023-12-09:
[image]
I guess the bridge layer isn't the problem. It's the layer right above the bridge that is the problem.

closed 2023-12-09 by Downquark7